SpacehuhnTech/esp8266_deauther #750: Request Packet Monitor and Deauther Detector

ChandraOrbit, 2018-03-27:
1. Versi 2.0 no Packet Monitor on WEB GUI just on OLED
2. Request .. If possible the DeauterDetector can also be merged with ESP8266_Deauther.

thanks

marek26340, 2018-03-27:
1. that's normal
2. it's already done, you just need that OLED display.

tobozo, 2018-03-27:
Packet Monitor in the Web GUI: it's [easy to do with u8glib](https://github.com/tobozo/ghetto_blaster/blob/8951443766fd16c71006be466571921a15a67ac2/wificontrols.h#L63) and shouldn't be a problem to achieve with the current library.

The real underlying problem is the load on the ESP.  Adding too much polling to the webserver increases the probability it'll crash.

The only way I can think about to have screencap without impacting the stability would be to send the 1Kb along with every JSON response the server does, and let JavaScript draw something out of it.

spacehuhn, 2018-03-27:
The problem is that you can't have an access point open and host a web server while sniffing WiFi.
It's like with every other WiFi card. Either you use it as an access point to host a network, as a station to connect to one or you put it in the monitor mode to sniff for packets. But you can't have everything at the same time.

tobozo, 2018-03-27:
doh!

spacehuhn, 2018-03-27:
Using 2 ESPs could solve that ;)

tobozo, 2018-03-27:
yep, keeping the idea for when I'll manage to flash an ESP8266 from an ESP32 :-)

ChandraOrbit, 2018-03-27:
I agree if there are 2 ESPs in 1 board, so the esp8266_deauther software feature, packet monitoring and deauther detection can be explored, the container is concise and does not reduce user portability as well as convenience.

lspoplove, 2018-03-27:
After I upgrade all my products I can consider designing such board  with 2 ESPs

发自Naver邮箱应用软件

spacehuhn, 2018-03-28:
The problem with 2 ESPs is, how would they communicate?
Via serial? That will make problems when trying to flash them! Unless there is some on/off switch for the serial lines.
Better would be a i2c or spi connection but I never programmed something like that, so no idea how much work that would be.

tobozo, 2018-03-28:
I have only tried serial so far, with partial success.

SPI version (untested yet): https://github.com/JiriBilek/WiFiSpiESP
Serial version (untested too):  https://github.com/jeelabs/esp-link

spacehuhn, 2018-04-04:
I'm going to close this because the main question got answered. 
It isn't possible to add the packet-monitor or deauth-detector to the web interface just like that. You'd need custom hardware and modified software, a lot of work for something that already works without problems using a $3 display from China.

closed 2018-04-04 by spacehuhn