SpacehuhnTech/esp8266_deauther #751: OLED board V3 messed up button control

mc-philis, 2018-03-27:
Hello, I've updated my DSTIKE V3 board with your last build, it work fine, was a little scared by the first black screen but it's just an option to activate the OLED through web-ui.

The issue is that the momentary slide switch﻿ is messed up, 

if I press the down button it's for ok
up button is for down
ok button is for up

Thanks for your work spacehuhn and dstike.

mc-philis, 2018-03-27:
Update, I just realised I might have flash the regular build and not the Oled V3 built. 
I'm trying again with this one and let you know.

spacehuhn, 2018-03-27:
I just tested the .bin file for the v3 board and it works :)
https://github.com/spacehuhn/esp8266_deauther/releases

If you used the Arduino code you need to adjust the button settings.
There is a configs folder where you can find the predefined config files for the DSTIKE boards.
That should be good for your board:
https://github.com/spacehuhn/esp8266_deauther/tree/master/configs/DSTIKE_Deauther_OLED_v2-v3
You have to take the A_config.h file and paste it in the Arduino sketch (esp8266_deauther) folder to replace the default A_config.h.

mc-philis, 2018-03-27:
It was my fault, with the proper OLEDV3 build it's perfect and oled is activated by default. 

I was flashing the default build.

closed 2018-03-27 by spacehuhn