SpacehuhnTech/esp8266_deauther #753: AP scan does not find any acces points after scan

sieuwe1, 2018-03-28:
Hi 

I am having a problem with the AP scan option. When i press the AP scan button first off all the light on the ESP8266 won't turn on. After i wait 20sec i reload the page and it won't show any networks. Is it possible it is not scanning at all because the blue light is not burning? When i do SSD beacon attack the blue light turns on so the LED is working. 

Hope someone can help. 

Greetings Sieuwe

Version i am running is 2.0 
Browser chrome on Android

spacehuhn, 2018-03-28:
Thanks.
I can see in the serial output `ERROR: Saving scan results in /scan.json`. Maybe something is corrupted in the file system. It happens to me from time to time.  
In the serial monitorry typing in `format` and press enter. Be sure you have it on `Newline` and `115200`. Then restart the board.

sieuwe1, 2018-03-28:
Hi 

I typed format and it formatted some files. But after that i gave me a bunch of errors. So i re flashed it using the Arduino IDE and now it works again. I guess there indeed was a corrupt file.

If you are having this problem:
1: type format in Serial monitor and restart the board
2: if this does not work re flash the board.

This should fix the problem.

Greetings 

Sieuwe

closed 2018-03-28 by sieuwe1