SpacehuhnTech/esp8266_deauther #754: Cannot connect to device using Windows 10

HtheB, 2018-03-28:
Somehow, my connection gets rejected when trying to connect to the board.
(used board is: esp8266)
Connecting on a phone does work, but somehow doesn't allow me to connect on my Windows 10 laptop...
used bin fw is v2.0

tobozo, 2018-03-28:
> (used board is: esp8266)
Please add more precision to this obvious information (i.e. post a pic of your board).

Also document yourself on the differences between ad hot and infrastructure WiFi connections, the problem you're mentioning here isn't in the deauther but in the way the WiFi connection is setup.

HtheB, 2018-03-28:
@tobozo I'm trying to connect to the pwned network. Like I said, it only works on my phones, not on my notebook.

Both devices gives the same results
(pics of used devices attached)
[image]

spacehuhn, 2018-03-28:
Sounds like a problem with your notebook or the operating system on your notebook.

tobozo, 2018-03-28:
Thanks, that's a Wemos Mini D1 and an ESP8266 Dev Kit.

The deauther works out of the box with any system, if you can connect from your mobile and not from your notebook, then check the settings on the notebook as it may be possible your WiFi card is dual band but the 2.4GHz band is blacklisted or flagged as unsafe.

spacehuhn, 2018-03-28:
Could also be that the laptop doesn't like to connect to a (local) network without internet connection.

HtheB, 2018-03-28:
@tobozo hmmm, that might be a reason... I should check it out.

@spacehuhn I can connect to my mobile hotspot without internet connection, so that shouldn't be the problem.

HtheB, 2018-03-28:
Ok, I've checked the logs, and the log says that it does not get any reaction of the router/apn
Nothing more, nothing less... -_-

It connects to all my other routers and apn's just fine..

tobozo, 2018-03-28:
[maybe this is related](https://windowsreport.com/fix-intel-ac-7260-driver/)

other random things to try:

- check if your other routers/apn's are using 5GHz band (will confirm the 2.4GHz is blacklisted)
- change the channel of the AP on the deauther
- don't put the ESP too close to the netbook (give it 1 meter) to minimize interferences
- [Capture the negociation with Wireshark](https://www.wireshark.org/download.html)

HtheB, 2018-03-28:
@tobozo 
Well, I can connect to the 5GHz of my router, but also 2.4GHz (I'm connected to the 2.4GHz one now)
I've tried Wireshark, but since I can't connect to the pwned APN, it doesn't work :(

HtheB, 2018-03-28:
I recorded my screen to show what I mean with 'cannot connect'.
Sorry, my Windows is in Dutch language. But I show you that I can connect to my own router just fine, but I can't to pwned :(
https://youtu.be/MAB7DzQByrY

tobozo, 2018-03-28:
hard to say what happens behind the UI :-)

But Windows 10 will also let you connect to WiFi from the command line, maybe it'll be more verbose on the reason it can't connect?

https://www.windowscentral.com/how-manage-wireless-networks-using-command-prompt-windows-10

Creogenesis, 2020-10-21:
My best ever laptop - Thinkpad T420 :) can't connect to pwnet on Windows, Linux Manjaro. WTF
It is not driver problem

Bommlon, 2020-10-26:
Are you able to connect from a different device?