SpacehuhnTech/esp8266_deauther #756: [CZ] Czech Language Support

JosefKuchar, 2018-03-28:
I've created Czech language support for the web interface (PR #755 ). I'm not 100% sure with some special cases, so if you have any suggestions you can write them down.

mgx0, 2018-05-22:
suggestions:

"setting_ssid" : "SSID přístupového bodu AP (je-li zapnuto).\nDélka musí být mezi 1 a 31 znaky",
"setting_password": "Heslo přístupového bodu AP (je-li zapnuto).\nDélka musí být mezi 8 a 31 znaky",
"enable_random": "Zapnout náhodní režim",
"disable_random": "Vypnout náhodní režim",
"random_desc": "Zapnout generování náhodného SSID seznamu",
"setting_serialEcho": "Aktivuje odozvu (echo) pro každý řádek v sériové konzoli",
"settings_webSpiffs": "Zapnout SPIFFS pro všechny webové soubory",
"setting_deauthspertarget": "Kolik deauth paketů bude odesláno každému cíli",
"setting_beaconchannel": "Pokud aktivní, beacon pakety budou odesílány na odlišných kanálech",
"setting_probesPerSSID": "Kolik probe paketů bude odesláno k SSID",

tobozo, 2018-06-02:
#883