SpacehuhnTech/esp8266_deauther #757: Deauther doesnt work

jeremey2002, 2018-03-28:
Hii i have a question everything works fine but wen i start deauther it doesnt work
When i start it i says he send packages but my wifi doesnt go of or slower
Pleas help me im.using esp8266 and the latest version deauther

digmorepaka, 2018-03-28:
hmm weird, i guess deauthentication packets are suposed to slow the wifi down

Phoenix1747, 2018-03-28:
@digmorepaka They are not intended to slow down the wifi, but to get all the connected devices to disconnect.
@jeremey2002 Maybe you need to increase the packet rate?

digmorepaka, 2018-03-30:
i know

tobozo, 2018-03-30:
@Phoenix1747 on some iPhones and notebooks there are situations where the deauthing works but you can only feel a slowdown because the device can reconnect very fast, and the OS doesn't care.

Phoenix1747, 2018-03-30:
@tobozo Yeah, right but that's only like a symptom of the deauther ^^

sudipto3331, 2018-03-31:
I have the same problem here.
All are working fine. But after pressing start button packets are not being sent.

spacehuhn, 2018-04-01:
Have you clicked the reload button to refresh the displayed packet rate?

jeremey2002, 2018-04-01:
How you increase the packet rate?
Because on my old tablet its disconnect  the other devices not?

Lilwiggy, 2018-04-01:
If you put it in something metallic like a pringles can the range and lackey rate will sky rocket

Lilwiggy, 2018-04-01:
Packet rate*

Phoenix1747, 2018-04-01:
@Lilwiggy You can log into the web ui and change the packet rate in the settings.

sudipto3331, 2018-04-01:
@spacehuhn Yeah. I had done it already. But no packets are being sent. As a result targeted wifi's clients aren't disconnecting. Even I have increased the packets rate. But still it isn't Working. It is noted that I'am using NodeMcu. I was using deauther V1.6 previously.

Phoenix1747, 2018-04-01:
@sudipto3331 Just out of curiousity, what specific packet rate are you using?

sudipto3331, 2018-04-01:
@Phoenix1747 Now I am using the value 20

spacehuhn, 2018-04-01:
Without the serial output, we can only speculare why that is.

sudipto3331, 2018-04-02:
With speculating, give some solution

tobozo, 2018-04-02:
With solution, close this issue ?

closed 2018-04-02 by tobozo