SpacehuhnTech/esp8266_deauther #759: Who is "sort" ?! Can someone help me?! (v2.0 on Wemos D1miniPRO)

TheMiguelBi, 2018-03-28:
Hello everyone!
I'm trying to load version 2.0, but an error blocks me.
With version 1.6 no problem.
I'm using:
- Wemos D1 mini PRO
- Oled 1.3" I2C display (D1 SDA and D2 SCL)
- 3 buttons (D5, D6, D7).

I am using Arduino Ide 1.8.3
What am I doing wrong?

The error is this:
___________________________________
sketch/Accesspoints.cpp: In member function 'void Accesspoints::sort()':
Accesspoints.cpp:8: error: 'class LinkedList<Accesspoints::AP>' has no member named 'sort'
   list->sort([](AP & a, AP & b) -> int{
         ^
sketch/Accesspoints.cpp: In member function 'void Accesspoints::sortAfterChannel()':
Accesspoints.cpp:16: error: 'class LinkedList<Accesspoints::AP>' has no member named 'sort'
   list->sort([](AP & a, AP & b) -> int{
         ^
exit status 1
'class LinkedList<Accesspoints::AP>' has no member named 'sort'
___________________________________
Thanks for any lighting!!

tobozo, 2018-03-28:
Speculation: another version of the library LinkedList was previously installed.
This is the version you need: https://github.com/ivanseidel/LinkedList

TheMiguelBi, 2018-03-28:
Thank you!! Now I managed to load the program successfully! 
Display and keys work!
Beacon Attack is okay, but I find some problems with the Deauth. 
Now I will do some tests.
Thanks again for the speedy solution!

closed 2018-04-04 by spacehuhn