SpacehuhnTech/esp8266_deauther #760: Boot loop after initial SPIFFS format.

kosso, 2018-03-28:
Hi,
I'm in a boot loop after flashing. 
Using a NodeMCU style ESP8266 board. 0.96" I2C OLED, RGB LED and 3 buttons. 

any ideas? 
thanks. 

~~~
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
...etc.
~~~

kosso, 2018-03-28:
Closing. Cause was dodgy pin definitions in the config.

closed 2018-03-28 by kosso

spacehuhn, 2018-03-28:
Yes the pins can make problems and the crashes won't tell you why. I spent hours debugging things like this.
I hope the default config wasn't the problem? If so let me know what I can improve :)

kosso, 2018-03-28:
Well, my particular issue was defining pins as `1`, `2`, `3`, etc.. and not `D1`, `D2`, `D3` ! 

/facepalm

spacehuhn, 2018-03-28:
Oh yes you have to use the D* pins or translate them into the actual GPIO pin numbers.  That's why I put a pinout of the NodeMCU in the setup guide in the wiki.

Anyway good to see you figured it out. I hope this helps some future readers too.