SpacehuhnTech/esp8266_deauther #762: Deauther 2.0: NodeMCU - timeout loading file run?...

careyer, 2018-03-30:
Hi there,

The Scan AP and Scan Stations buttons do not work for me. After a while the following timeout message comes up and Deauther seems to have halted. I can't switch beteen the different pannels anymore (Scan, SSID, Attack...)

[image]

BTW: On Firefox is seems to run an inital scan automatically (SSIDs and signal strenth are shown) whereas on IE and Chrome  the list of results stays empty.

tobozo, 2018-03-30:
You can get more debug info by using the serial connection.

If you can reproduce this, it would be helpful to copy/paste the serial output.

careyer, 2018-03-30:
Sure... willing to help! How can I open a serial connection? Putty? Via Wifi? Via COM? Bausrate?

tobozo, 2018-03-30:
Putty can do, but you may have to install a driver to see the ESP as a COM port 
Baudrate is 115200

schinfo, 2018-03-30:
It looks like there is a bug - in A_config.h (Line 48) setting "#define USE_DISPLAY false // will force display on" doesn't work - display is always on. I've the same problems and can reproduce it  - maybe it's a timing problem when the display is on. I've reported it to Spacehuhn - waiting for his answer.

tobozo, 2018-03-30:
@schinfo maybe create an issue for that ?

Also don't forget to power-cycle the ESP after changing the display settings, if you only reset the ESP, the display will "remember" its last settings and can stay on.

tobozo, 2018-03-30:
@careyer it seems like a normal behaviour, it can be annoying if you have more than one pre-recorded SSID+credentials, this is why the Serial interface comes more handy :-)

Please close this issue if it's solved.

spacehuhn, 2018-03-30:
Scan aps should work, if not paste in the serial output.
Scan stations will cause a hold the access point until the scan is finished - you will have to reconnect.

tobozo, 2018-04-02:
closing this as solved, feel free to reopen if applicable

closed 2018-04-02 by tobozo

vitaliyivanov0802, 2020-06-28:
> Привет,
> 
> Кнопки Scan AP и Scan Station не работают для меня. Через некоторое время появляется следующее сообщение таймаута, и Deauther, кажется, остановился. Я больше не могу переключать beteen на разные панели (сканирование, SSID, атака...)
> 
> [image]
> 
> BTW: на Firefox, кажется, запустить начальное сканирование автоматически (SSIDs и сигнал strenth показаны), в то время как на IE и Chrome список результатов остается пустым.

Как подключишь устройство к сети напряжения, включи на своем телефоне не вайфай, а точку доступа.
Как погаснет синяя лампочка - включаешь wi-fi, подключается к pwned, и вауаля, работает как новый.