SpacehuhnTech/esp8266_deauther #763: #define USE_DISPLAY false" doesn't work

schinfo, 2018-03-30:
It looks like there is a bug - in A_config.h (Line 48) setting "#define USE_DISPLAY false // will force display on" doesn't work - display is always on (also after power ON/OFF).

tobozo, 2018-03-30:
Pasting this here as it appears to have been posted in the wrong issue:

@shinfo Wrote: 
It looks like there is a bug - in A_config.h (Line 48) setting "#define USE_DISPLAY false // will force display on" doesn't work - display is always on. I've the same problems and can reproduce it - maybe it's a timing problem when the display is on. I've reported it to Spacehuhn - waiting for his answer.

tobozo, 2018-03-30:
have you tried to comment out `// #define USE_DISPLAY` instead of assigning `false` ?

schinfo, 2018-03-30:
@tobozo 

It's no possible - USE_DISPLAY must be declared!

tobozo, 2018-03-30:
Just had a look at the code, and it seems like it's more a technical debt than a bug.

I only could find the [signature](https://github.com/spacehuhn/esp8266_deauther/blob/63eb971b137f5e15443c51dce8fa79e0fb443c6d/esp8266_deauther/Settings.h#L80) but no traces of the implementation.

So setting display to true or false  has probably the same effect as yelling at the ESP (or maybe yell louder?).

spacehuhn, 2018-03-30:
6f88113ffbe1875261aa5fb64b41b2c62450d5ee

closed 2018-04-04 by spacehuhn