SpacehuhnTech/esp8266_deauther #764: Au

martytee, 2018-03-30:
Hoping you can help that I am receiving this error in Arduino.
-------------------------------------------------------
C:\ESP8266 Deauther\v2.0\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:10:0:
sketch\functions.h:5:25: fatal error: ArduinoJson.h: No such file or directory

 #include "ArduinoJson.h"

                         ^

compilation terminated.

Using Library EEPROM at version 1.0 in folder: C\Users\Marty_001\AppData\Local\Arduino15\packages\esp8266\2.0.0\libraries\EEPROM
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
--------------------------------------------------------------

spacehuhn, 2018-03-30:
Have you read and followed the installation guide?
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

tobozo, 2018-03-30:
Make sure you select the ESP8266 from the "Deauther Modules" group in the boards list.
If it isn't there, follow the Installation guide again.

martytee, 2018-03-30:
Thanks for getting back so quick and believe I followed what you sent, but am now getting the following error:

---------------------------------------------------------------

Arduino: 1.8.5 (Windows 8.1), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 921600, 4M (3M SPIFFS)"

In file included from sketch\Settings.h:8:0,

                 from C:\ESP8266 Deauther\v2.0\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:12:

sketch\A_config.h:7:24: fatal error: SH1106Wire.h: No such file or directory

 #include "SH1106Wire.h"

                        ^

compilation terminated.

Using library EEPROM at version 1.0 in folder: C:\Users\marty_001\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\EEPROM 
Using library ArduinoJson at version 5.13.1 in folder: C:\Users\marty_001\Documents\Arduino\libraries\ArduinoJson 
Using library Wire at version 1.0 in folder: C:\Users\marty_001\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\Wire 
Using library SPI at version 1.0 in folder: C:\Users\marty_001\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\SPI 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

---------------------------------------------------------------

tobozo, 2018-03-30:
The missing library is supposed to come with the SDK 2.0.0, but if it fails to load you can still install it from the library manager, look for "SSD1306" and download the one from David Eichhorn.
Same could apply to LinkedList.

[image]

Reasons why it wouldn't load:
- Arduino environment was modified after SDK 2.0.0 install (i.e. changed the library/project path in the preferences.txt)
- There is a space in the path name of your project that could trigger random behaviours on compilation
- The path to the SDK 2.0.0 should contain "packages/deauther" and not "packages/esp8266", see what it gives on my system: 
`Using library EEPROM at version 1.0 in folder: /home/tobozo/.arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/EEPROM `
- You're running the deauther 2.0 source on the old environment (version 1.6 where the ArduinoJSON and LinkedList don't exist)

martytee, 2018-03-30:
Big thx tobozo & spacehuhn, turned out that I didn't look past my nose and had a path error.  Great Work!!!

closed 2018-03-30 by tobozo

matchu20002, 2018-06-15:
thanks for the link. yes it does sound really similar. This is my file path:

Users/matthewdodd/Downloads/esp8266_deauther-master 2/esp8266_deauther/esp8266_deauther.ino

I guess its a case of trying to figure out what needs to change to make it fix the path error.

Also I'm using Board -  Generic ESP8266 Module. The port is cu.Bluetoorh-Modern.