SpacehuhnTech/esp8266_deauther #765: Upload Problem

FloidHekmatyar, 2018-03-30:
Hello guys :) 

i tried too upload the Code bug i get the following error message. 

In file included from C:\Users\Jonas\Desktop\Wifi\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino:10:0:

sketch\functions.h:5:25: fatal error: ArduinoJson.h: No such file or directory

 #include "ArduinoJson.h"

                         ^

compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 Module.

spacehuhn, 2018-03-30:
Duplicate of https://github.com/spacehuhn/esp8266_deauther/issues/764 ?

spacehuhn, 2018-04-04:
Were you able to fix it?

tobozo, 2018-04-09:
Closing this issue for lack of followup, feel free to reopen if anything new comes up.

closed 2018-04-09 by tobozo