SpacehuhnTech/esp8266_deauther #766: Use Arduino IDE build for esp8285 web server "FileNotFound"

zanjie1999, 2018-03-31:
I use Arduino IDE to build esp8266_deauther for esp8285 
I connect to wifi "pwned" and use browser to open 
http://192.168.4.1
web server return "FileNotFound"

tobozo, 2018-03-31:
On ESP8266 this problem is resolved by using the Sketch Data Uploader, but also the file should be copied on first run si I guess something went wrong.

Can you post the output of the serial when the ESP is booting ?

zanjie1999, 2018-03-31:
Mounting SPIFFS...ERROR
ERROR couldn't open /settings.json
Settings loaded from /settings.json
true/settings.json
ERROR couldn't open /names.json
Device names loaded from /names.json
ERROR couldn't open /ssids.json
SSIDs loaded from /ssids.json
ERROR: Saving scan results in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0
Executing /autostart.txt
Done executing script

tobozo, 2018-03-31:
can you try the command "format" and see if the output gets better ?

spacehuhn, 2018-03-31:
`Mounting SPIFFS...ERROR`
In the upload/compile settings, did you gave it room for using the SPIFFS? Like `1mb (256kb)` is a good configuration. If you set it to 512kb only however, the SPIFFS is disabled because everything is used for the program memory and it won't work.

zanjie1999, 2018-03-31:
I flash "ESP8266_Deauther_v2.0.0_1MB.bin" to esp8285

serial output:

Mounting SPIFFS...ERROR
ERROR couldn't open /settings.json
Copying /web/attack.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/index.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/info.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/scan.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/settings.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/ssids.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/style.css.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/attack.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/scan.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/settings.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/site.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/ssids.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/lang/de.lang.gz from PROGMEM to SPIFFS...ERROR
Copying /web/lang/en.lang.gz from PROGMEM to SPIFFS...ERROR
Copying /web/LICENSE.gz from PROGMEM to SPIFFS...ERROR
Settings loaded from /settings.json
true/settings.json
Copying /web/attack.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/index.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/info.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/scan.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/settings.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/ssids.html.gz from PROGMEM to SPIFFS...ERROR
Copying /web/style.css.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/attack.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/scan.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/settings.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/site.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/js/ssids.js.gz from PROGMEM to SPIFFS...ERROR
Copying /web/lang/de.lang.gz from PROGMEM to SPIFFS...ERROR
Copying /web/lang/en.lang.gz from PROGMEM to SPIFFS...ERROR
Copying /web/LICENSE.gz from PROGMEM to SPIFFS...ERROR
ERROR couldn't open /names.json
Device names loaded from /names.json
ERROR couldn't open /ssids.json
SSIDs loaded from /ssids.json

tobozo, 2018-03-31:
Just compiled it for you using @spacehuhn recommended settings:

- [ESP8266_deauther_v2.0.0_ESP8285_._1M_256K.bin](https://github.com/spacehuhn/esp8266_deauther/releases/download/v2.0/ESP8266_deauther_v2.0.0_ESP8285_._1M_256K.bin)

zanjie1999, 2018-03-31:
OK, thank!

Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.1
Executing /autostart.txt
Done executing script