SpacehuhnTech/esp8266_deauther #772: Deauther 2.0: can't access the web

ifengchao, 2018-03-31:
After flashing the firmware with esp8266 flash tool, i can type command via the serial interface, but can't access the web. V1.6 is working good.

Here are some serial log:

  Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Copying /web/attack.html.gz from PROGMEM to SPIFFS...OK
Copying /web/index.html.gz from PROGMEM to SPIFFS...OK
Copying /web/info.html.gz from PROGMEM to SPIFFS...OK
Copying /web/scan.html.gz from PROGMEM to SPIFFS...OK
Copying /web/settings.html.gz from PROGMEM to SPIFFS...OK
Copying /web/ssids.html.gz from PROGMEM to SPIFFS...OK
Copying /web/style.css.gz from PROGMEM to SPIFFS...OK
Copying /web/js/attack.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/scan.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/settings.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/site.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/ssids.js.gz from PROGMEM to SPIFFS...OK
Copying /web/lang/de.lang.gz from PROGMEM to SPIFFS...OK
Copying /web/lang/en.lang.gz from PROGMEM to SPIFFS...OK
Copying /web/LICENSE.gz from PROGMEM to SPIFFS...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.0
Executing /autostart.txt
Done executing script
[AP] request: /hotspot-detect.html NOT FOUND
[AP] request: / OK
[AP] request: /style.css OK
[AP] request: /web/lang/en.lang OK
[AP] request: /scan.html OK
[AP] request: /style.css OK
[AP] request: /scan.html OK
[AP] request: /style.css OK
[AP] request: /apscan.html NOT FOUND
[AP] request: / OK
[AP] request: /style.css OK
[AP] request: / OK
[AP] request: /generate_204 NOT FOUND
[AP] request: /gslb/ NOT FOUND
[AP] request: /rsp204 NOT FOUND
[AP] request: /rsp204 NOT FOUND
[AP] request: /rsp204 NOT FOUND
[AP] request: /wifi/cw.html NOT FOUND
[AP] request: /ws/awaken/page/ NOT FOUND

ifengchao, 2018-03-31:
Well, it worked after reboot.
The first boot really takes more time than the v1.6 ?

ifengchao, 2018-03-31:
The web loading cost too much time, and often connect time out, seems it's not quite stable?

tobozo, 2018-03-31:
Apparently your phone has some applications in desperate need of a WiFi connection and this generates more load on the ESP, confusion ensues :p 

Tips: 
- Firefox doesn't do that many phoney requests.
- Private browsing sessions offer a different behaviour (no cache) from the regular sessions.

spacehuhn, 2018-04-01:
The first boot takes time because of the file copying you see in the serial log.
The web loading times can be improved by using a private/incognito window.
The NOT FOUND is normal, that is your device asking for stuff from the webserver. 
I can also confirm that Firefox on Linux  doesn't do this kind of thing. Windows 10 however likes to ask for a connecttest. txt.

spacehuhn, 2018-04-04:
First of all: Great new issue label!
Secondly: I updated the web server. If you can, download the repository and upload the code using Arduino. It should give some performance improvements - it won't be perfect, but in my tests it performed way faster.

ifengchao, 2018-04-09:
Thank you guys, i'll test it later when i got time.

tobozo, 2018-04-29:
@ifengchao any improvements after the last release?

tobozo, 2018-05-12:
closing this as inactive, feel free to reopen it this still occurs on the latest version

closed 2018-05-12 by tobozo