SpacehuhnTech/esp8266_deauther #773: How Do i setup Deautherization on macOS High Sierra?

Meksel, 2018-04-02:
i didnt get the esp8266, yet and i want to make sure i can set it up on mac is that possible and also if i do get esp8266 i am going to get this "ESP8266 CP2102" is that a good/acceptable one?

spacehuhn, 2018-04-02:
Everything you need to know should be in the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki) :)
Every ESP8266 will work, some differencies are explained. Drivers are also linked and there is a flasher for Mac too, if you don't like to compile it yourself with Arduino.

Meksel, 2018-04-02:
Ahh, So i just need to download the mac version of the flasher where i put all the files into the esp8266 and is that it?

jLynx, 2018-04-03:
You can follow along with http://deauth.me/ (site should be up in the next day or so)

r0073rr0r, 2018-04-03:
esptool.py doing good for me 🗡

Meksel, 2018-04-04:
@jLynx Nice website, will it tell me what to do and then i can deauther my internet?

jLynx, 2018-04-04:
@Meksel right now it only shows you how to install it. But when I have time I can add to it so it shows you how to use it

tobozo, 2018-04-09:
@Meksel did all your questions get anwers? If so, can you close this issue?

tobozo, 2018-04-21:
Closing this as inactive

closed 2018-04-21 by tobozo