SpacehuhnTech/esp8266_deauther #778: Web Site Issue

keremdikmen, 2018-04-04:
hi, I uploaded the code any error.  But when i try use web site it didn't anything. Only i can see interface and surfing menus, except that didnt work any buttons. Otherwise i could write scan, reset, reboot etc. on serial port monitor. 
   I added serial interface when i was try web site. 
https://yadi.sk/i/sOKzGveE3U4viU

tobozo, 2018-04-04:
Try to issue this command in the serial monitor:

`format`

Then reset the ESP manually and observe what's being written in the serial monitor, be patient :-)

Explanation: on first boot, the ESP attempts to copy the webfiles onto the SPIFFS, this can take some time, if the copy is interrupted (reset, reboot, power cycle) then you only get a partial web content and erratic behaviour. Formatting the SPIFFS is the only way to trigger the automatic copy again.

Also are you using a browser in incognito mode ? From the 404 log on your screen capture, the .js files are called, but the .js.gz files should be served instead.

keremdikmen, 2018-04-04:
Thanks for reply. I've formatted before you say. And now It's same result again. I tried incognito mode on my phone (Chrome). Nothing changed
[image]

spacehuhn, 2018-04-04:
Ok, try this:
Type in `format` then press the reset button. Copy the **whole** serial output that comes up when the device restarts. 
Also type in `get settings` and paste the output. I did some changes to the web-server code recently. Maybe I can find the error with those infos.

keremdikmen, 2018-04-04:
I have done this. Is it right?
[image]

tobozo, 2018-04-04:
`webSpiffs: false` :thinking:

spacehuhn, 2018-04-04:
Found the error. 
Ok 2 ways to fix it for you. Type in `set webSpiffs true` then `save settings` and restart.
Or upload the updated Code. I just pushed the changes to the repo.

Explaination: The webserver uses the SPIFFS (a file system on the memory chip) but that turned out to make problems for some users, so I made it optional with the `webSpiffs` setting. Now if that setting is on false (which it is by default), it will serve the web files directly from the programm memory. When it is on true, it will copy everything onto the SPIFFS and serve the files from there. That programm-memory web server had some bugs and didn't react to all requests correctly. I'm working on it now.

keremdikmen, 2018-04-05:
I just writed `set webSpiffs true` and reboot. It's work properly now. Thank you very much.

tobozo, 2018-04-11:
@spacehuhn can we close this or are you still working on the resolution ?

spacehuhn, 2018-04-11:
It's fixed in the current code :)

closed 2018-04-11 by spacehuhn