SpacehuhnTech/esp8266_deauther #779: Upload problem(failed to send 36 bytes)

MonkeysHack, 2018-04-04:
What should I do, this is what it says when I try and upload it please respond.

Archiving built core (caching) in: C:\Users\KAIDA~1.KAI\AppData\Local\Temp\arduino_cache_797629\core\core_deauther_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_1M256_d52fae24e609dfb18e59525ff1db9f0a.a
Sketch uses 654666 bytes (85%) of program storage space. Maximum is 761840 bytes.
Global variables use 42120 bytes (51%) of dynamic memory, leaving 39800 bytes for local variables. Maximum is 81920 bytes.
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tobozo, 2018-04-04:
This sounds like a bad USB cable but it could also be a generic flashing problem (not deauther related).

Please specify the device you're using and how it's wired when flashing, eventually post a pic of your build so we can see the wiring.

MonkeysHack, 2018-04-04:
I'm using a nodemcu not sure which version it is, and I tried two other cable but it still didn't work.

tobozo, 2018-04-05:
Eventually push the flash button while Arduino is compiling and keep it pushed until you see it's uploading.

On older versions of nodemcu you had to do some ninja thing before flashing:
- push reset (keep it pushed)
- push the flash button (keep it pushed)
- click on "upload" in the Arduino IDE (ninja skill or third hand required)
- release the reset button

tobozo, 2018-04-06:
closing this as resolved, thanks for reporting the solution!

closed 2018-04-06 by tobozo

devimeruva, 2019-01-19:
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0

 this is my error and it's repeating. I think my drivers have no problem but i am stuck here. Can you please help me ?