SpacehuhnTech/esp8266_deauther #781: LinkedList<MenuNode> problem

peszabolcs, 2018-04-04:
I has got the following problem message:

Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Alaplap:"Generic ESP8266 Module, Serial, 160 MHz, 80MHz, DIO, 115200, 512K (64K SPIFFS), nodemcu"

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:153: error: 'class LinkedList<MenuNode>' has no member named 'size'

       else currentMenu->selected = currentMenu->list->size() - 1;

                                                       ^

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:172: error: 'class LinkedList<MenuNode>' has no member named 'size'

       if (currentMenu->selected < currentMenu->list->size() - 1) currentMenu->selected++;

                                                      ^

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:218: error: 'class LinkedList<MenuNode>' has no member named 'get'

         if(currentMenu->list->get(currentMenu->selected).hold) currentMenu->list->get(currentMenu->selected).hold();

                               ^

DisplayUI.cpp:218: error: 'class LinkedList<MenuNode>' has no member named 'get'

         if(currentMenu->list->get(currentMenu->selected).hold) currentMenu->list->get(currentMenu->selected).hold();

                                                                                   ^

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:230: error: 'class LinkedList<MenuNode>' has no member named 'get'

           if( currentMenu->list->get(currentMenu->selected).click){

                                  ^

DisplayUI.cpp:231: error: 'class LinkedList<MenuNode>' has no member named 'get'

             currentMenu->list->get(currentMenu->selected).click();

                                ^

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:479: error: no matching function for call to 'LinkedList<MenuNode>::remove(uint8_t&)'

       apListMenu.list->remove(apListMenu.selected);

                                                  ^

sketch\DisplayUI.cpp:479:50: note: candidate is:

In file included from sketch\DisplayUI.h:5:0,

                 from sketch\DisplayUI.cpp:1:

sketch\LinkedList.h:42:6: note: void LinkedList<T>::remove(T) [with T = MenuNode]

 void LinkedList<T>::remove(T value) {

      ^

sketch\LinkedList.h:42:6: note:   no known conversion for argument 1 from 'uint8_t {aka unsigned char}' to 'MenuNode'

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:520: error: no matching function for call to 'LinkedList<MenuNode>::remove(uint8_t&)'

       stationListMenu.list->remove(stationListMenu.selected);

                                                            ^

sketch\DisplayUI.cpp:520:60: note: candidate is:

In file included from sketch\DisplayUI.h:5:0,

                 from sketch\DisplayUI.cpp:1:

sketch\LinkedList.h:42:6: note: void LinkedList<T>::remove(T) [with T = MenuNode]

 void LinkedList<T>::remove(T value) {

      ^

sketch\LinkedList.h:42:6: note:   no known conversion for argument 1 from 'uint8_t {aka unsigned char}' to 'MenuNode'

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:552: error: no matching function for call to 'LinkedList<MenuNode>::remove(uint8_t&)'

       nameListMenu.list->remove(nameListMenu.selected);

                                                      ^

sketch\DisplayUI.cpp:552:54: note: candidate is:

In file included from sketch\DisplayUI.h:5:0,

                 from sketch\DisplayUI.cpp:1:

sketch\LinkedList.h:42:6: note: void LinkedList<T>::remove(T) [with T = MenuNode]

 void LinkedList<T>::remove(T value) {

      ^

sketch\LinkedList.h:42:6: note:   no known conversion for argument 1 from 'uint8_t {aka unsigned char}' to 'MenuNode'

sketch\DisplayUI.cpp: In lambda function:

DisplayUI.cpp:569: error: no matching function for call to 'LinkedList<MenuNode>::remove(uint8_t&)'

       ssidListMenu.list->remove(ssidListMenu.selected);

                                                      ^

sketch\DisplayUI.cpp:569:54: note: candidate is:

In file included from sketch\DisplayUI.h:5:0,

                 from sketch\DisplayUI.cpp:1:

sketch\LinkedList.h:42:6: note: void LinkedList<T>::remove(T) [with T = MenuNode]

 void LinkedList<T>::remove(T value) {

      ^

sketch\LinkedList.h:42:6: note:   no known conversion for argument 1 from 'uint8_t {aka unsigned char}' to 'MenuNode'

sketch\DisplayUI.cpp: In member function 'void DisplayUI::drawMenu()':

DisplayUI.cpp:712: error: 'class LinkedList<MenuNode>' has no member named 'size'

   else if (currentMenu->selected >= currentMenu->list->size()) currentMenu->selected = currentMenu->list->size() - 1;

                                                        ^

DisplayUI.cpp:712: error: 'class LinkedList<MenuNode>' has no member named 'size'

   else if (currentMenu->selected >= currentMenu->list->size()) currentMenu->selected = currentMenu->list->size() - 1;

                                                                                                           ^

DisplayUI.cpp:715: error: 'class LinkedList<MenuNode>' has no member named 'size'

   for (int i = row; i < currentMenu->list->size() && i < row + 5; i++) {

                                            ^

DisplayUI.cpp:716: error: 'class LinkedList<MenuNode>' has no member named 'get'

     tmp = currentMenu->list->get(i).getStr();

                              ^

sketch\DisplayUI.cpp: In member function 'void DisplayUI::clearMenu(Menu*)':

DisplayUI.cpp:792: error: 'class LinkedList<MenuNode>' has no member named 'size'

   while (menu->list->size() > 0) {

                      ^

DisplayUI.cpp:793: error: no matching function for call to 'LinkedList<MenuNode>::remove(int)'

     menu->list->remove(0);

                         ^

sketch\DisplayUI.cpp:793:25: note: candidate is:

In file included from sketch\DisplayUI.h:5:0,

                 from sketch\DisplayUI.cpp:1:

sketch\LinkedList.h:42:6: note: void LinkedList<T>::remove(T) [with T = MenuNode]

 void LinkedList<T>::remove(T value) {

      ^

sketch\LinkedList.h:42:6: note:   no known conversion for argument 1 from 'int' to 'MenuNode'

exit status 1
'class LinkedList<MenuNode>' has no member named 'size'

I don't know what to do right now and I will be really thankful if anyone know what I need to do.

spacehuhn, 2018-04-04:
Try using the latest code by cloning the repository.

The error is probably because you have the LinkedList library installed somewhere and it is an older version than used for the deauther. The SDK comes with the correct library installed but Arduino gives priority to the user libraries and by that, your LinkedList installation.
You can probably fix it by removing the library from your Arduino library folder or updating it. You can also download the latest code that works without needing the LinkedList library.

peszabolcs, 2018-04-04:
Thank you, it worked, but now, I've got an other message:

Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Alaplap:"Generic ESP8266 Module, Serial, 160 MHz, 80MHz, DIO, 115200, 512K (64K SPIFFS), nodemcu"

Felépítés beállítások megváltoztak, egész újraépítése
Archiving built core (caching) in: C:\Users\user\AppData\Local\Temp\arduino_cache_954491\core\core_esp8266_esp8266_generic_UploadTool_esptool,CpuFrequency_160,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_nodemcu_77ac0782251438eb4abb1ecd518a9c59.a
c:/users/user/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\user\AppData\Local\Temp\arduino_build_77353/esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

sketch\Names.cpp.o:(.text._ZNK11ArduinoJson11JsonVariant14variantAsArrayEv[_ZNK11ArduinoJson11JsonVariant14variantAsArrayEv]+0x8): undefined reference to `__cxa_guard_acquire'

sketch\Names.cpp.o:(.text._ZNK11ArduinoJson11JsonVariant14variantAsArrayEv[_ZNK11ArduinoJson11JsonVariant14variantAsArrayEv]+0xc): undefined reference to `__cxa_guard_release'

sketch\Names.cpp.o: In function `ArduinoJson::JsonArray::invalid()':

sketch/Names.cpp:297: undefined reference to `__cxa_guard_acquire'

sketch/Names.cpp:297: undefined reference to `__cxa_guard_release'

sketch\SSIDs.cpp.o: In function `ArduinoJson::JsonArray::invalid()':

C:\Users\user\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ArduinoJson\src/ArduinoJson/Serialization/JsonSerializerImpl.hpp:103: undefined reference to `__cxa_guard_acquire'

C:\Users\user\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ArduinoJson\src/ArduinoJson/Serialization/JsonSerializerImpl.hpp:103: undefined reference to `__cxa_guard_release'

sketch\SSIDs.cpp.o:(.text._ZN5SSIDs4loadEv+0x8d): undefined reference to `__cxa_guard_acquire'

sketch\SSIDs.cpp.o:(.text._ZN5SSIDs4loadEv+0x9e): undefined reference to `__cxa_guard_release'

sketch\Settings.cpp.o: In function `ArduinoJson::JsonObject::invalid()':

sketch/Settings.cpp:353: undefined reference to `__cxa_guard_acquire'

sketch/Settings.cpp:353: undefined reference to `__cxa_guard_release'

collect2.exe: error: ld returned 1 exit status

spacehuhn, 2018-04-04:
Are you sure you use the Deauther SDK?
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide

peszabolcs, 2018-04-04:
There aren't any option to ESP8266 Deauther Module... I've got only ESP8266
Module.

Stefan Kremser <notifications@github.com> ezt írta (időpont: 2018. ápr. 4.,
Sze 22:28):

marek26340, 2018-04-05:
I'm sure that he's not using the Deauther SDK...

tobozo, 2018-04-05:
@peszabolcs you can use it by involving yourself into following every step of the [installation documentation](https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide).

We can help but we won't assist.

peszabolcs, 2018-04-05:
Ohh I deleted, and restarted, and now, it's working!
So mutch thanks!

closed 2018-04-05 by peszabolcs