SpacehuhnTech/esp8266_deauther #782: EspExceptionDecoder

tfujdiler, 2018-04-05:
Hi, maybe you know how to fix it ? When I try to use Esp Exception Decoder in Arduino 1.8.5 I have a message : ERROR: xtensa-lx106-elf-gdb.exe not found!
[image]

tobozo, 2018-04-05:
Hello, it's a known problem, see [this issue](https://github.com/me-no-dev/EspExceptionDecoder/issues/34)

Since this problem is recent, using the [previous release](https://github.com/me-no-dev/EspExceptionDecoder/releases/tag/1.0.6) should temporarily do it.

I'll close this issue as it's not deauther related.
If you need a followup, just add a comment on the [issue](https://github.com/me-no-dev/EspExceptionDecoder/issues/34) so when a solution comes up you'll be notified.

closed 2018-04-05 by tobozo

tfujdiler, 2018-04-05:
If I use v1.0.6 mesage is:
ERROR: elf was not found!  :)
I would like to check for errors in deuther 1.6,
is there any other solution ?

spacehuhn, 2018-04-05:
The "elf" error usually shows up when you haven't compiled it yet. Press compile and open the decoder again.

tfujdiler, 2018-04-05:
Thanks guys, it works but I think I can not use it :(
Spacehuh deauther is amazing, thank you for creating it,
I see the great potential of this program :)
First I must get rid of errors
The problem is you can see it well in the monitor when there are two identical lines:
getting attacks JSON (267)...done
getting attacks JSON (267)...done
[image]
then on my smartphone I lose the server connection, 
the attack is still working, the diode is on
When smartphon lose connection with server monitor show me :

running Deauth attack... done
running Deauth attack... done
running Deauth attack... done
running Deauth attack... done
running Deauth attack... done
running Deauth attack... done

when I restart wifi on the phone,
even though the attacked networks have different 
channels everything works as it should 
on the monitor it looks like this :

getting attacks JSON (267)...done
running Deauth attack... done
running Deauth attack... done
getting attacks JSON (267)...done
running Deauth attack... done
running Deauth attack... done
getting attacks JSON (267)...done

the problem arises when I first start deauther, later it is stable
I do not know if it's a bug in the program or my Samsung Galaxy S6 is responsible for it
getting attacks JSON (xxx)...done
getting attacks JSON (xxx)...done
the smartphone loses connection to the server :(
guys please help me find a solution to this problem

### Hardware/Flashing:
Board:				 NodeMcuESP8266 V2 Amica
Using OLED:                     NO
Using WebServer:            ?YES?NO?
Using Serial:                     ?YES?NO?
Installation mode:           Arduino 1.8.5
Code/binary version:       esp8266_deauther_1mb(1).bin
Flash size:                        4M
Flash method:                 Node MCU FLASCHER
Flash Frequency:             80
Upload Speed:	       115200
Powered by:                   USB & PHONE CHARGER
Smartphone                   SGS6  (Android 7.0)
[image]

tobozo, 2018-04-05:
All this is a perfectly normal behaviour, unfortunately getting disconnected from the Web UI when scanning or attacking can't be avoided.

tfujdiler, 2018-04-05:
If this is a normal reaction when we start deauther, then I have one idea :)
We need a function which when an error is detected
getting JSON (xxx) attacks ... done
getting JSON (xxx) attacks ... done
will reset the wi-fi on the phone then the problem will become less nagging and everything will be better :) spacehuhn can you fix that ? :)