SpacehuhnTech/esp8266_deauther #785: Deauther not working

GastWLAN, 2018-04-05:
I have searched the first 10 pages of the issues but didnt find a solution. I have flashed the board(no oled) v2.0.0 and I can access the interface without problem. But when I attack a network it does nothing. 

I tried attacking from an iPhone and nothing. Also from android no results. When I attacked from my laptop the iPhone immediately disconected=sucess. But after that one time the iPhone stayed always connected. I managed after many attemps to disconnect it once more time but sience then not again.

I raised the packages to 50, I am getting 32 and more during the attack but no results.

Please help.

tobozo, 2018-04-06:
Maybe you iPhone has switched from 2.4GHz to 5GHz ? The deauther can still see it, but wont be able to attack 5GHz range.

GastWLAN, 2018-04-06:
I dont know how I can see this but in general the deauther didnt work at all apart from the 2 times it disconnected the iphone. Android phone and tablet and windows laptop not affected. I tried reflashing but no improvement. Any settings to change? Do the other boards with oled displays work better? Thanks for the help!

GastWLAN, 2018-04-06:
Ok thanks!

GastWLAN, 2018-04-06:
I switched to 2.4Ghz and it worked! But how about 5Ghz? There's no way to attack these networks?

tobozo, 2018-04-06:
An ESP8266 can only do 2.4GHz so no, you can't do that with the esp8266_deauther.
You may need specific hardware to do that, maybe the device you're using on your laptop is [compatible](https://www.aircrack-ng.org/doku.php?id=compatible_cards)?

spacehuhn, 2018-04-06:
About the 5GHz: https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ#5ghz-support

tobozo, 2018-04-09:
Closing this issue as the 2.4GHz part is resolved and 5GHz is out of the range of this project.

closed 2018-04-09 by tobozo

spacehuhn, 2022-02-06:
Hi, I know this issue is old and has been closed, but since it ranks well on Google, I want to use that chance to link to a new blog post that discusses this topic in further detail:
https://blog.spacehuhn.com/deauth-attack-not-working/

Jordenm77, 2022-03-23:
Do they make any better style antenna for the deauther watch I have the V2 currently

borbod, 2023-03-31:
Hi, I made this project with the oled ssd1306 display with the wemos board and I also appled the code on the menu and the modes are shown in the display, but the buttons for the top and bottom do not work the button is not broken with the wire I connected thank you (connections are absolutely true) thank you for helping .

ArticPolarDEV, 2023-11-01:
> Olá, fiz esse projeto com o display oled ssd1306 com a placa wemos e também apliquei o código no menu e os modos são mostrados no display, mas os botões de cima e de baixo não funcionam o botão não está quebrado com o fio que conectei obrigado (as conexões são absolutamente verdadeiras) obrigado por ajudar.

Same error, bro.