SpacehuhnTech/esp8266_deauther #786: English Language Updates

M-rTom, 2018-04-06:
Attacks page; has "loose" instead of "lose" at the top.
Settings page; Beaconchannel has "will sent" instead of "will send".

About all I can find. :)

tobozo, 2018-04-06:
Thanks for reporting this, but looking at the history it seems [it's already been fixed](https://github.com/spacehuhn/esp8266_deauther/commit/3dd1969850a791c90c33729026acc7733133ddcb).

How old is your version ?

spacehuhn, 2018-04-06:
Fixed the "will sent" now :)

M-rTom, 2018-04-07:
Ack. I downloaded the 2.0 binary file to make it easy for my first test run. I haven't compiled anything myself yet, so I didn't check to see if that was already repaired. Thank you everyone for your work in this project.