SpacehuhnTech/esp8266_deauther #790: How to fix it ?

tfujdiler, 2018-04-06:
Hi, I found smal bug in APs when SSID it's too long and does not contain spaces btn,
the layout and size of the rest of the page change.
Expo_InternetDomowyM10/33 this name is one string of characters in addition, it does not contain a dash.
You have an idea how to change it ?
[image]

tobozo, 2018-04-06:
[image]

tfujdiler, 2018-04-06:
Changing the text on the button is a last resort :)
sooner I will go to the indicated address M18 / 33 and try to log in to the neighbor router 
maybe "admin admin" work :) then i can change his SSID :)
If this option does not work I will knock on the door but i use the hammer :)
and I will ask why it has such a long network name ;)

tobozo, 2018-04-06:
technically a SSID can go up to 32 chars so the UI should be able to display the complete string, or at least not break the layout when it's overflowed.

[Actually it's already half implemented](https://github.com/spacehuhn/esp8266_deauther/blob/master/web_interface/style.css#L238), the missing part is in the `table-layout: fixed;` in the [table declaration](https://github.com/spacehuhn/esp8266_deauther/blob/master/web_interface/style.css#L232):

    table {
        table-layout: fixed;
        width: 100%;
        min-width: 400px;
        margin-bottom: 2em;
    }

can you try that and see if it fixes the word-break problem ?

if it can save the life of your arrogant neighbour's router it's worth a try !

tfujdiler, 2018-04-06:
I was close I changed the width to 80%, 
[image]
first impression great the table has decreased and the neighbor's network has disappeared :) but when I refreshed the list of networks the table was again on the entire width of the screen and the M10 / 33 network appeared :( 
I have the impression that the length of the network name forces the width of the table
I will try use your idea but file style.css is different in the deauther 1.6 version

tfujdiler, 2018-04-06:
I change my mind :) I will not be buried in style.css I suffered a lot of failures for one day,
so time to be bad :)
will direct all available power esp into this network with a long name,
maybe it will trigger self-ignition of the router ;)
and the problem will be solved by the flames :) anyway thanks for the help :+1: 
I'm glad that the support works :)

tfujdiler, 2018-04-06:
I think the topic can be moved to closed ones ;)

[image]

closed 2018-04-11 by tobozo