SpacehuhnTech/esp8266_deauther #791: error: espcomm_sync failed.

mrchichu, 2018-04-06:
hi all, i tried to uploading the code to board. however it failed
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

im using board: wemos d1 https://www.ebay.com/itm/WeMos-D1-CP2102-CH340G-ESP8266-USB-NodeMcu-Lua-V3-Internet-Wifi-Develop-Board-/202108353172

how can i solve this problem?

i did check the port. 
i did install the board.

tobozo, 2018-04-06:
you can check the cable
you can check the driver

spacehuhn, 2018-04-06:
https://github.com/spacehuhn/esp8266_deauther/wiki/Troubleshooting

ThunderGod1991, 2018-04-07:
I am having the same problem and I am using the adafruit huzzah I tried uploading other sketches with the esp8266 board selected and i have no issues but when I select the board with esp8266 deauther modules  i get an error I have also tried different upload speeds but nothing works

spacehuhn, 2018-04-07:
Also see: https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#installation-tips-and-tricks

ThunderGod1991, 2018-04-07:
I am using the arduino ide could it be a driver issue if i don't have trouble uploading other sketches when i select the board under the esp8266 modules?

spacehuhn, 2018-04-07:
In that case the problem is with the upload settings. They are all explained in the wiki, I send the links. Try to play around with them. Not every board works with the same settings.

ThunderGod1991, 2018-04-07:
Okay thank you for helping me I will try changing the settings

ThunderGod1991, 2018-04-07:
I tried different settings and no luck does the deauther program support the cp2104 driver on the huzzah board?

tobozo, 2018-04-09:
@ThunderGod1991 you said:

> when I select the board with esp8266 deauther modules i get an error

what error do you get exactly ? just changing boards in the menu should not raise any error at all

ThunderGod1991, 2018-04-09:
I can upload example codes and my own written codes just fine when i have selected the board under the esp8266 modules but as soon as i select the board under the esp8266 deauther options and try to upload the deauther code I get this error

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

tobozo, 2018-04-09:
I see 9 boards under the esp8266_deauther tab, which one are you selecting? what are the settings?

Recommended failsafe settings:
- board: Generic Esp8266 Module
- flash mode: DIO
- flash size: 1M(256KSPIFFS)
- Reset method => try "nodemcu" if "ck" doesn't work

ThunderGod1991, 2018-04-09:
I was using the adafruit huzzah board option but i tried the generic esp8266 option with the nodemcu reset method and it works thanks for helping me

tobozo, 2018-04-09:
So the root issue is there's probably a problem with the huzzah board profile.

I take note that the same huzzah profile in the SDK 2.3.0 compiles other sketches all right for this board, am I right?

tobozo, 2018-04-09:
I compared the huzzah boards.txt part between sdk 2.0.0 and sdk 2.3.0 and found something.

Can you look for `Arduino/packages/deauther/hardware/esp8266/2.0.0-deauther/boards.txt`,  delete those three lines, restart the Arduino IDE and see if it uploads?

```
huzzah.menu.UploadTool.esptool=Serial
huzzah.menu.UploadTool.esptool.upload.tool=esptool
huzzah.menu.UploadTool.esptool.upload.verbose=-vv
```

These lines have been removed from the boards.txt bundled with the sdk 2.3.0 so they're probably the cause of the error messages you had earlier.

mrchichu, 2018-04-10:
okay i have changed the port. may i know howi can check to sure that the number of pin i entered in code match the pin on the board?

tobozo, 2018-04-18:
Just refer to the [pinout of your board](https://cdn-learn.adafruit.com/assets/assets/000/046/249/original/adafruit_products_Huzzah_ESP8266_Pinout_v1.2-1.png?1504885873).

Closing this as solved, feel free to reopen if you still get the ` error: espcomm_open failed` message.

closed 2018-04-18 by tobozo