SpacehuhnTech/esp8266_deauther #794: Serial: DELAY function has unspecified unit (and it is not seconds)

dglaude, 2018-04-08:
In serialcommands.md it is not specified the unit used for "DELAY" function.

> DELAY
> delay <time>
> Will pause the serial command interface for a given time.

It does not seems to be seconds (I tested with 60 and it was not effective).
But it seems to work (I tested with 60000 and I was locked out, I had to reset).

SCAN is explicitly in second (for -t and -c)
ATTACK -t is unspecified
RUN is explicitly in second (for -c)

spacehuhn, 2018-04-09:
The unit is in milliseconds as known from the Arduino delay() function:  
`DELAY 1000`
You can also directly give it seconds:  
`DELAY 1s`

dglaude, 2018-04-09:
I am just saying that it is not documented (like for ATTACK).

Also some scripting command are in seconds and other in millisecond. It could help if all were using the same unit for time parameter.

tobozo, 2018-04-21:
@spacehuhn do we need a SETUNIT [h|m|s|ms|us] command to solve that or can we close this as wontfix ?

spacehuhn, 2018-04-21:
wontfix.
You can use `s` and `ms` for every time parameter. I added the info the the delay command.
There should be no confusion.

closed 2018-04-21 by tobozo