SpacehuhnTech/esp8266_deauther #795: Wrong Deauth Packet

mlg556, 2018-04-08:
This will sound weird, but when I select and attack my iPhone (with MAC 2c:f0:a2:98:37:35) connected to my network MLG74, (with BSSID f8:23:b2:71:a8:75); the deauth packets sent from the ESP module appear to be coming from another unrelated wrong AP, (with BSSID c0:25:e9:95:7b:00) even though my iPhone is not and can not be connected to that AP. It would seem that the program is "misreading" the access-point the client is connected to (maybe an index problem?). I have attached to image files explaining the problem. I have also emptied the module and re-uploaded the .ino file, which did not seem to fix it.

[image]
[image]

spacehuhn, 2018-04-09:
Thanks! I will try to reproduce the error and look into it.

mlg556, 2018-04-09:
Thanks for the reply! I have managed to identify the problem further. The `scan` correctly identifies the stations and their access points, but when I run an empty deauth attack`attack -d` (with nothing selected) and then prompt `show` to show the list one more time (without scanning again), the devices and their AP's are all wrong. It would seem that somehow the `attack -d` code messes up with the list. I have attached two images illustrating this. Also, it is important to note that when I `add`my device to the list with a name, select it with that name and attack; the attack successfully works.
[image]
[image]

spacehuhn, 2018-04-09:
Yes the problem is when the AP-list gets sorted and the indices change. I'm currently restructuring the code to fix that.

spacehuhn, 2018-04-09:
Should be fixed now

mlg556, 2018-04-09:
Thx for the quick fix !

tobozo, 2018-04-09:
@mlg556 thanks for the quality of your feedback

spacehuhn, 2018-04-09:
Yeah thanks for the detailed infos! We're not seeing that very often here ;D

closed 2018-04-11 by spacehuhn