SpacehuhnTech/esp8266_deauther #799: no continuous scanning mode

ErfanDL, 2018-04-10:
Hi. where is continuous scanning mode in version 2.0 ? how to enable ?

thanks

spacehuhn, 2018-04-10:
That is possible over the serial interface, see the reference: https://github.com/spacehuhn/esp8266_deauther/blob/master/serialcommands.md

spacehuhn, 2018-04-10:
`set led false` and then
`save settings` and restart the device

ErfanDL, 2018-04-10:
@spacehuhn thanks the led is disabled.  I connecting neopixel rgb to pin SD2 but nothing happen and cant turning on.

tobozo, 2018-04-21:
@ErfanDL did you manage to get your neopixel ring to turn on?

closed 2018-04-21 by tobozo

ErfanDL, 2018-04-21:
@tobozo 
No. neopixel cannot working. I have a problem with compiling from source. after compile and modifying code to working with neopixel the board cannot boot and stock at bootloop.