SpacehuhnTech/esp8266_deauther #800: can not boot !

ErfanDL, 2018-04-10:
Hi. after compiling from source nodemcu cannot booting up and show tis messages in seriam monitor. any fix ?

```
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

spacehuhn, 2018-04-10:
Have you tried to upload the reset sketch: https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch
and flash the deauther again?

ErfanDL, 2018-04-11:
@spacehuhn 
yes I format esp8266 flash and reprogram but cant fix.

tobozo, 2018-04-11:
can you please post a pic of your ESP so we can see what model it is ?

ErfanDL, 2018-04-11:
@tobozo 
the model is Nodemcu 

[image]

spacehuhn, 2018-04-11:
Does it maybe work without all the pins connected?

ErfanDL, 2018-04-11:
@spacehuhn
No not working ! but the .bin pre compiled working without any problem. I want to compile it from source

spacehuhn, 2018-04-11:
If it works works with the bin file I would suspect an error in your configuration (A_config.h) or the upload settings.

tobozo, 2018-04-29:
@ErfanDL did you get your neopixel working? Please keep us posted about your turing path

ErfanDL, 2018-04-29:
@tobozo
my Nodemcu cannot booting up when compiling from source and I can't test neopixel. I got boot loop with the below message

```
Mounting SPIFFS...OK
Formatting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

Mar10us, 2018-05-10:
I had the exact same problem with the exact same NodeMCU board. The binary works fine but i wanted to use the NeoPixel instead of the onload LED. I tried to compile it with the pin for the NeoPixel on the default GPIO9, but it keeps rebooting like your board.

I fixed it by changing the pin for the NeoPixel LED from GPIO9 to GPIO1.

FYI: 
I used the following settings to flash:
- Board: Generic ESP8266 Module  
- Flash Mode: DIO
- Flash Frequency: 80 MHZ
- CPU Frequency: 160 MHz
- Flash Size: **1M (256K SPIFFS)**  or **4M (3M SPIFFS)** (i tried both)
- Reset Method: nodemcu
- Upload Speed: 115200

tobozo, 2018-05-10:
@ErfanDL can you confirm this workaround is valid on your build?

spacehuhn, 2018-06-28:
Closing due to inactivity.

closed 2018-06-28 by spacehuhn