SpacehuhnTech/esp8266_deauther #801: Beacon...SSID...o_O

TheMiguelBi, 2018-04-10:
Deauther 2.0.3
on WEMOS D1 Mini
-------------------

I noticed during the Beacon attack an anomaly on the SSID name.
Once the SSIDs of my home wifi network have been cloned, I have deleted the WPA key on 3 of the 60 cloned networks. Once the changes were saved, I started the attack.
However, the 3 networks have also undergone a change to the name. Anything after a " " space is deleted. I believe the spaces are indigestible for the creation of new SSIDs. Or maybe I'm neglecting some details ?!
Here a short video demonstration: https://youtu.be/sO5xLTaCHpM

I just wanted to point out this, nothing serious ...

spacehuhn, 2018-04-11:
Hm... 
do the SSIDs also appear wrong on the list after the change (after you click reload)?

TheMiguelBi, 2018-04-11:
Yes, right. 
You can not see it in the video, but if you change the name and save it to the SSIDs window, the name also changes to the list after the page update.

spacehuhn, 2018-04-11:
Should be fixed now 😃 
aef4c985345adac73ad4c25636debbdd59d0123e

TheMiguelBi, 2018-04-11:
I've just uploaded version 2.0.4, 
but still the problem of space "eat" letters is present. : /

spacehuhn, 2018-04-12:
The old js file is probably still cached in your browser, that's why.

TheMiguelBi, 2018-04-12:
Everything perfect!! 
It was like you said. Thank you!
(I tried from another device, and everything works).

closed 2018-04-12 by TheMiguelBi