SpacehuhnTech/esp8266_deauther #802: No AP is shown, attacking via Serial interface works

JsBergbau, 2018-04-11:
With this https://www.aliexpress.com/item/1PCS-Wireless-module-CH340-NodeMcu-V3-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/32721337812.html and https://github.com/spacehuhn/esp8266_deauther/releases/download/v2.0/ESP8266_Deauther_v2.0.0_1MB.bin deauther works.

With that https://www.aliexpress.com/item/WAVGAT-NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-flash-USB-serial-CH340G/32841025867.html no WIFI-AP pwned is shown. However connecting via serial works, typing scan and than attack -da brings all devices offline. Any idea why? Is this node MCU damaged? 

Interesting to notice: Second node MCU (working only via serial) can send out 120 packets/s, correctly working one only 60 (or 80). How comes this difference?

tobozo, 2018-04-11:
have you tried with the [4MB binary](https://github.com/spacehuhn/esp8266_deauther/releases/download/v2.0/ESP8266_Deauther_v2.0.0_4MB.bin) ?

JsBergbau, 2018-04-11:
Just tried, exactly the same behaviour.

tobozo, 2018-04-11:
If you can capture the serial output after it's been flashed (and while you're using it), there may be some clues about why the webfiles aren't there

JsBergbau, 2018-04-11:
``▒▒d`{▒▒o▒▒l ▒▒'▒
Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.0
Executing /autostart.txt
Done executing script

Thats the output after a reset, captured at 115200

spacehuhn, 2018-04-11:
hm.. looks like everything is correct 😕

JsBergbau, 2018-04-11:
> Also how far is your mobile from the nodemcu ? Maybe the 'optimized' onboard antenna has something to do with it ?

It is about 1 Meter away. 5 cm away it is still not found.

Tried the 4 MB reset sketch, still no AP shown. Output of the reset sketch

`sd�c��'o�$g'���cp��dsdsd8�o��$��b'�|$����c��og��ogl o;���'��d`{ۓo��$`��g�
STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initializedSPIFFS cleaned
WiFi credentials erased
DONE!
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

STARTING...SPIFFS cleaned
WiFi credentials erased
DONE!

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initialized

EEPROM initialized
EEPROM cleaned
SPIFFS initialized
`

The second node MCU is currently not available. I've compared the RSSI with my Galaxy
Node MCU
0 XXXX                                    6  -29 WPA2 XXXXXX Avm
 1 XXX                                     6  -83 WPA2 XXXX AvmAudio

Galaxy
0 -45 dBm
1 -74 dBm

They are right beside each other and because of HF effects, they compare quite well.

The Wifi-Module seems to be ok, otherwise deauth and scanning would not work.

schinfo, 2018-04-11:
Please, update to the new version. You use an old one!

STARTED! \o/
v2.0.0         <======================= newest is v2.0.3 !!!!
Executing /autostart.txt

spacehuhn, 2018-04-11:
"old" is relative here. But yes maybe try uploading the latest code with Arduino. I made some changes to the web server, who knows, maybe it fixes your problem. 
You can also try uploading one of the example sketches Arduino provides you. They have an example which hosts an access point. If that also doesn't work, it's definitely a hardware problem.

schinfo, 2018-04-11:
Also your board has 32MB flash - so maybe it's a problem.
Try to change the Flash size in the settings of the Arduino IDE.

schinfo, 2018-04-11:
Try setting "4M (1M SPIFFS)"

JsBergbau, 2018-04-11:
Now I've compiled:

Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.1
Executing /autostart.txt
Done executing script

But it is now worse. RSSI (same position as before is now -54 dBm instead of -31 dBm as before, other AP isn't found anymore).

If played a bit with the flashsize and captured output shortly after flashing, perhaps it helps

;ld▒▒|▒$▒|
▒
$▒c|ǃ▒{▒c▒
c▒▒o'▒lno▒▒▒
bx▒▒l;l{l8▒g▒
▒l
Ĝ
b
'▒|$▒▒▒▒c▒▒o'▒l▒▒d`▒gl`o;Ǜ▒'
▒▒d`{ۓo
▒▒$`▒▒g▒
Mounting SPIFFS...OK

Soft WDT reset

ctx: cont
sp: 3fff1780 end: 3fff1b80 offset: 01b0

>>>stack>>>
3fff1930:  40002514 3fffdd3c 00000000 000020b0
3fff1940:  00000100 3fff3090 00000008 00000000
3fff1950:  00000000 000020c0 00000000 00000000
3fff1960:  00000000 3fff1bb0 3fff32a0 40205970
3fff1970:  3fff20f5 3fff1990 00000004 4010178e
3fff1980:  00000000 3fff3090 0000ffff 00000000
3fff1990:  3fff1a10 00000001 00000000 3fffbff0
3fff19a0:  00000000 00000000 3fff4960 00000000
3fff19b0:  3fff32a0 4010695e 3fff1a80 00000001
3fff19c0:  3fff1a10 00000001 3fff1bb0 40202cc9
3fff19d0:  3fff32a0 00000000 3fff1bb0 40202f17
3fff19e0:  3fff32a0 00000001 3fff1a80 4025bab4
3fff19f0:  3ffe91d4 3fff1a80 00000001 40225c18
3fff1a00:  3ffe91d4 000001dd 3fff1a30 402278fd
3fff1a10:  3ffe00ff 3fff1a80 3fff1a30 402278d5
3fff1a20:  3fff4900 3fff1a80 00000001 4021e192
3fff1a30:  3ffe9258 00000000 000003e8 4022876e
3fff1a40:  3fff4960 3fff49b0 3fff1a70 40228870
3fff1a50:  3ffe91d4 006e6f73 3fff1ae0 40228894
3fff1a60:  3fff48f0 3fff1ad0 3fff1ae0 4021f251
3fff1a70:  3fff4940 0000000f 0000000e 402286e6
3fff1a80:  00000000 0000209f 0000209f 4022873e
3fff1a90:  3fff1a9c 000001dd 3fff1ae0 4022876e
3fff1aa0:  3fff1ba4 3fff0774 3fff1ae0 40228870
3fff1ab0:  40101a6d 3fff1af0 3fff0774 4025bab4
3fff1ac0:  3fff1ba4 3fff0774 3fff0774 4021aa29
3fff1ad0:  3ffe8728 40232c8d 00000000 00000fa0
3fff1ae0:  3fff4900 0000000f 0000000e 3fff4900
3fff1af0:  0000000f 0000000e 3fff5c70 000001df
3fff1b00:  000001dd 000000a6 000000a7 000003ff
[... 81 lines cut ...]
Copying /web/settings.html.gz from PROGMEM to SPIFFS...OK
Copying /web/ssids.html.gz from PROGMEM to SPIFFS...OK
Copying /web/style.css.gz from PROGMEM to SPIFFS...OK
Copying /web/js/attack.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/scan.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/settings.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/site.js.gz from PROGMEM to SPIFFS...OK
Copying /web/js/ssids.js.gz from PROGMEM to SPIFFS...OK
Copying /web/lang/de.lang.gz from PROGMEM to SPIFFS...OK
Copying /web/lang/en.lang.gz from PROGMEM to SPIFFS...OK
Copying /web/LICENSE.gz from PROGMEM to SPIFFS...OK
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.1
Executing /autostart.txt
Done executing script

-----

With the official v2.0.0 1MB version I saw for some seconds the wifi pwned, but could not connect. After a few seconds it was gone, showed up for some seconds and disappeared for a longer time. Any ideas?

tobozo, 2018-04-11:
wut three attempts before the files are finally copied?

I you have the ESP Exception decoder in the Arduino /tools menu it's worth pasting there the stack dump, but it'll only work after you compile the sketch.

It would also be insightful to see how this NodeMCU behaves with a [regular WiFi scan](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiScan/WiFiScan.ino) compiled from the 2.3.0 SDK, and compare the behaviour with the other ESP.

If you get the same differences, that will only opt-out the SDK2.0.0 as a possible source of the problem, but it'll also opt-in the NodeMCU as possibly faulty.

mgx0, 2018-04-13:
no, it's not visible on any of devices, tried 3 iphones, 3 Macs, one Dell laptop, none of them showed the SSID. This happens on two NodeMCU V3 boards from WEMOS. Now I'm trying LoLin board (bigger than the WEMOS) and I can see the SSID now.
Power source is good in both cases.

any idea what can cause this? I have seen some situations where AP was not visible when specific MAC address range has been used for AP. But I don't see an option to change MAC address via serial, is it possible? just to try this

tobozo, 2018-04-13:
time to get more insights on your hardware: https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples/CheckFlashConfig

mgx0, 2018-04-13:
sure, here's the output:

`Flash real id:   001640C8
Flash real size: 4194304 bytes

Flash ide  size: 1048576 bytes
Flash ide speed: 80000000 Hz
Flash ide mode:  DIO
Flash Chip configuration wrong!`

tobozo, 2018-04-13:
seems like a regular 4M/1M memory footprint

have you tried to use the sketch data uploader from the esp8266_deauther folder ? most web files are in there so it may spare the ESP from having to copy them all on first boot, and maybe offer a workaround to the AP not being started

mgx0, 2018-04-13:
you mean upload the content of the web_interface folder to SPIFFS?

tobozo, 2018-04-13:
gz is what the AP is sending to your browser anyway, the other folder is for maintenance, its content is minified & gzipped every time a new language is added or some changes are done to the web ui

Now oustide the deauther, do you have a working example of your ESP doing a basic WiFi scan ?
What's the RSSI difference between both models ?

mgx0, 2018-04-13:
here it is. I removed SSIDs from the list and ordered it to easier comparison

here is the working model:

```
     1  -56 WPA2 18:d6:c7:64:5b:22 Tp-LinkT
     6  -74 WPA* ac:22:05:29:a1:3e CompalBr
    11  -74 WPA* 00:90:a9:09:37:ab WesternD
     2  -75 WPA2 70:8b:cd:cc:7a:50 AsustekC
     2  -75 WPA* 70:8b:cd:cc:7a:51 AsustekC
     2  -75 WPA2 70:8b:cd:cc:7a:52 AsustekC
    11  -77 WPA2 00:4f:6a:0a:5a:78
     6  -80 WPA* 90:5c:44:a0:6a:ab CompalBr
     6  -81    - 92:5c:14:a0:6a:ab
    11  -84  WPA c8:3a:35:29:ee:48 TendaTec
```

here is the non-working model:

```
     1  -55 WPA2 18:d6:c7:64:5b:22 Tp-LinkT
     6  -74 WPA* ac:22:05:29:a1:3e CompalBr
    11  -79 WPA* 00:90:a9:09:37:ab WesternD
     2  -76 WPA2 70:8b:cd:cc:7a:50 AsustekC
     2  -76 WPA* 70:8b:cd:cc:7a:51 AsustekC
     2  -77 WPA2 70:8b:cd:cc:7a:52 AsustekC
    11  -83 WPA2 00:4f:6a:0a:5a:78
     6  -79 WPA* 90:5c:44:a0:6a:ab CompalBr
     6  -75    - 92:5c:14:a0:6a:ab
     1  -88  WPA c8:3a:35:29:ee:48 TendaTec
```

very small difference in my opinion. what would a difference mean?

tobozo, 2018-04-13:
A big difference in RSSI could mean broken or defective antenna but it looks like that's not the problem either.

mgx0, 2018-04-14:
No, these boards were used for another projects as clients, antenna is not a problem. But give me some time, I'll write a short sketch to act as AP and we'll see whether the general API has problems or it's just this one

mgx0, 2018-04-14:
Update: the 1.6 version of deauther works without problems on the same board which does not work with v2. Also custom sketch which creates AP works

Urdinat, 2018-04-16:
I have a similar problem. The board is chinese verison of Wemos D1 P1. When I run the sisinfo command from the console, I got this result:
# sysinfo
[======== SYSTEM INFO ========]
RAM usage: 51856 bytes used [64%], 30064 bytes free [36%], 81920 bytes in total

Current WiFi channel: 1
AP MAC address: 00:00:00:00:00:00
Station MAC address: 5c:cf:7f:68:0b:04
SPIFFS: 2761 bytes used [0%], 954553 bytes free [99%], 957314 bytes in total
        block size 8192 bytes, page size 256 bytes
Files: 
  /names.json 2 bytes
  /ssids.json 27 bytes
  /autostart.txt 0 bytes
  /settings.json 514 bytes
  /scan.json 24 bytes
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
===============================

If I understand correctly, the problem of the board's visibility and the ability to connect to it in the parameter "AP MAC address: 00:00:00:00:00:00"
One time instead of 00: 00: 00: 00: 00 it contained a real address, then the board was visible from the phone and it was possible to connect to it.
In addition, feature request: I really want to be able to configure the properties of the access point through the serial port.
And I apologize for the machine translation.

JsBergbau, 2018-04-18:
Just testet with the NodeMCU from https://de.aliexpress.com/item/WAVGAT-NodeMCU-V3-Lua-WIFI-modul-integration-von-ESP8266-zus-tzlichen-speicher-32-Mt-USB-serial/32841025867.html
As statet AP isn't shown with Version 2.0.0, but works fully via serial.
Other with @Urdin-at sysinfo shows 2 MAC-Adresses, one for AP und one for Station, so this is not the problem and it has 2 different, as it should.

```
# sysinfo
[======== SYSTEM INFO ========]
RAM usage: 54024 bytes used [66%], 27896 bytes free [34%], 81920 bytes in total

Current WiFi channel: 1
AP MAC address: de:xxxxxxxxxx
Station MAC address: dc:xxxxxxxx
SPIFFS: 28363 bytes used [0%], 2920887 bytes free [99%], 2949250 bytes in total
        block size 8192 bytes, page size 256 bytes
Files:
  /web/attack.html.gz 1441 bytes
  /web/index.html.gz 1022 bytes
  /web/info.html.gz 1425 bytes
  /web/scan.html.gz 1209 bytes
  /web/settings.html.gz 880 bytes
  /web/ssids.html.gz 1213 bytes
  /web/style.css.gz 1607 bytes
  /web/js/attack.js.gz 394 bytes
  /web/js/scan.js.gz 1978 bytes
  /web/js/settings.js.gz 623 bytes
  /web/js/site.js.gz 1078 bytes
  /web/js/ssids.js.gz 940 bytes
  /web/lang/de.lang.gz 3019 bytes
  /web/lang/en.lang.gz 2779 bytes
  /web/LICENSE.gz 687 bytes
  /names.json 2 bytes
  /ssids.json 27 bytes
  /autostart.txt 0 bytes
  /settings.json 478 bytes
  /scan.json 24 bytes
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
===============================

```

I've also tested v1.6, AP is shown flawlessly. However when an attack is started via Webinferface, it doesn't work anymore and thus Attack can only be stopped by pulling the power plug. The attack itself works.

Hope this helps to track down the problem.

tobozo, 2018-05-12:
please test against the latest version an reopen if necessary

closed 2018-05-12 by tobozo

JsBergbau, 2018-06-08:
Hallo,

just tested with ESP8266_Deauther_v2.0.5_4MB.bin and this NodeMCU https://de.aliexpress.com/item/WAVGAT-NodeMCU-V3-Lua-WIFI-modul-integration-von-ESP8266-zus-tzlichen-speicher-32-Mt-USB-serial/32841025867.html

No AP is shown. Scanning and attacking via serial works, so same behaviour as before.

mgx0, 2018-06-08:
already answered, we all have a fake board. load the 1.6, that one works even on our fake boards