SpacehuhnTech/esp8266_deauther #803: Error when uploading with Arduino IDE

Sly031, 2018-04-12:
Hi all,

I am new to all this. Just got my NodeMCU today, and bought it just for the wifi deauther. I had no luck flashing with ESP8266 flasher so tried uploading with Arduino IDE. I followed all the steps here: [https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#compiling-using-arduino-ide](url)

When clicking upload I get this error:

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 9600, 512K (64K SPIFFS), ck"

sketch\DisplayUI.cpp: In member function 'void DisplayUI::setupDisplay()':

DisplayUI.cpp:16: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]

   display.setFont(DejaVu_Sans_Mono_12);

                                      ^

In file included from C:\Users\maury\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display/SH1106Wire.h:31:0,

                 from sketch\A_config.h:7,

                 from sketch\Settings.h:8,

                 from sketch\DisplayUI.h:10,

                 from sketch\DisplayUI.cpp:1:

C:\Users\maury\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display/OLEDDisplay.h:187:10: error:   initializing argument 1 of 'void OLEDDisplay::setFont(const char*)' [-fpermissive]

     void setFont(const char *fontData);

          ^

Multiple libraries were found for "SH1106Wire.h"
 Used: C:\Users\maury\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display
 Not used: C:\Users\maury\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-4.0.0
 Not used: C:\Users\maury\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-4.0.0
 Not used: C:\Users\maury\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-4.0.0
 Not used: C:\Users\maury\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-4.0.0
exit status 1
**_invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]_**

Please help

spacehuhn, 2018-04-12:
You have to install the latest version of the display library: https://github.com/ThingPulse/esp8266-oled-ssd1306
Or delete the old, so that the library in the SDK takes its place.  

Reference: https://github.com/spacehuhn/esp8266_deauther/issues/749

tobozo, 2018-04-18:
closing this as duplicate, please refer and followup on #749 instead

closed 2018-04-18 by tobozo