SpacehuhnTech/esp8266_deauther #806: xpected initializer before 'DejaVu_Sans_Mono_12'

mgx0, 2018-04-13:
Hello, 

I'm getting this error, not sure why. I just downloaded the project, board definitions and this one is the only error I'm getting

```
In file included from sketch/DisplayUI.cpp:1:0:
DisplayUI.h:80: error: expected initializer before 'DejaVu_Sans_Mono_12'
unsigned const uint8_t DejaVu_Sans_Mono_12[] PROGMEM = {
                        ^
sketch/DisplayUI.cpp: In member function 'void DisplayUI::setupDisplay()':
DisplayUI.cpp:18: error: 'DejaVu_Sans_Mono_12' was not declared in this scope
   display.setFont(DejaVu_Sans_Mono_12);
                   ^
exit status 1
expected initializer before 'DejaVu_Sans_Mono_12'
```

mgx0, 2018-04-13:
seems the 'unsigned' keyword causes this error, after removal it compiles

tobozo, 2018-04-13:
yep it's the same issue as #749 
I've added a label to this issue, closing this as solved

closed 2018-04-13 by tobozo