SpacehuhnTech/esp8266_deauther #807: files ESP8266WiFi.cpp and ESP8266WiFi.h arent showing up and neither is the SDK_fix folder

xxsurajbxx, 2018-04-14:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).  
Please put error messages and code ```inside these 3 quotes/grave accents```
(Delete this text when you've read it)

spacehuhn, 2018-04-14:
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation

closed 2018-04-14 by spacehuhn

xxsurajbxx, 2018-04-15:
] I got this error message while trying to compile the esp8266_deauther.ino
(#808)
I ended up using my Mac instead and didn't get that problem. Instead I keep
getting the error which says that espcomm_open failed. I tried holding down
the flash button then pressing upload, and waiting until the arduino ide
finishes compiling and says uploading then I released the button. This is
what it says on the website and it doesn't work for me. Btw for some reason
my Mac doesn't show com 1 or any other coms, I linked a picture of my
ports. I also linked a video of me trying to upload the code. (I cut the
video in half and sent 2 videos because it was to big to send as a whole