SpacehuhnTech/esp8266_deauther #808: I got this error message while trying to compile the esp8266_deauther.ino

xxsurajbxx, 2018-04-14:
Arduino: 1.8.5 (Mac OS X), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 512K (64K SPIFFS), ck"

sketch/DisplayUI.cpp: In member function 'void DisplayUI::setupDisplay()':
DisplayUI.cpp:11: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]
   display.setFont(DejaVu_Sans_Mono_12);
                                      ^
In file included from /Users/Suradaawesomest/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/esp8266-oled-ssd1306-4.0.0/src/SH1106Wire.h:34:0,
                 from sketch/A_config.h:7,
                 from sketch/Settings.h:8,
                 from sketch/DisplayUI.h:10,
                 from sketch/DisplayUI.cpp:1:
/Users/Suradaawesomest/Library/Arduino15/packages/deauther/hardware/esp8266/2.0.0-deauther/libraries/esp8266-oled-ssd1306-4.0.0/src/OLEDDisplay.h:201:10: error:   initializing argument 1 of 'void OLEDDisplay::setFont(const uint8_t*)' [-fpermissive]
     void setFont(const uint8_t *fontData);
          ^
exit status 1
invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

TheMiguelBi, 2018-04-15:
The problem is the library "esp8266-oled-ssd1306". Try a previous version. 3.2.7
Which OLED display are you using?

spacehuhn, 2018-04-15:
You have to install the latest version of the display library: https://github.com/ThingPulse/esp8266-oled-ssd1306
Or delete the old, so that the library in the SDK takes its place.  

Reference: https://github.com/spacehuhn/esp8266_deauther/issues/749

tobozo, 2018-04-18:
closing this as duplicate, please refer and followup on #749 instead

closed 2018-04-18 by tobozo