SpacehuhnTech/esp8266_deauther #810: Whitelist registered MAC address and Deauth Everyone Else.

darkprince1337, 2018-04-17:
Hello.
I'm working on a project on WIFI security, i have to debar/deauth unauthorized users from our WIFI network.
I've created captive portal with radius authentication on a raspberrypi 3. It uses MAC auth but it's not reliable in terms of security.
So i want to add NodeMCU ESP8266 module to deauth users which are not registered (their MAC address) to my database. 
Can you help me out?

spacehuhn, 2018-04-17:
Please read this:
https://github.com/spacehuhn/esp8266_deauther/wiki/Serial#deauth-all

marek26340, 2018-04-17:
You don't need to use a deauther to just congest the 2.4GHz band more than it's really necessary. If you will need to make a whitelist of devices that can be on the network and then block everyone else, then use the MAC filtering function of the router. 
Why should I/you pay for a feature that's free and included in one device? :D
At least I hope that you didn't buy a very simple and cheap router without any features *sigh*

darkprince1337, 2018-04-18:
Thanks for reply.
I'm using ESP8266 because it's an Hardware based project.

Can i e-mail you my problem statement? have a look if you got some time. It will be really helpful for me to discuss it with you.

marek26340, 2018-04-18:
In most cases, you need some software to be able to work with some hardware. I still don't understand your project, so, if you will send some more details, it would be helpful, but not for fixing the issue... But how can you get my email without me sharing it on the entire internet (e.g. writing email here)?

tobozo, 2018-04-21:
closing this as the discussion moved to another channel, don't forget to share the results of your experiments :+1:

closed 2018-04-21 by tobozo

SansarSah, 2022-06-10:
@darkprince1337  did you find the solution?