SpacehuhnTech/esp8266_deauther #818: 2.0.0 wifi client

bilogic, 2018-04-23:
Hi,

Does 2.0.0 have wifi client capabilities? Thank you.

spacehuhn, 2018-04-23:
not yet, no :(

bilogic, 2018-04-24:
I see.. it worked in earlier versions of deauther. what I did was to:

1. flash wi-pwn v9.0 and enable the wifi client
2. flash deauther
3. wifi client remains functional

but this was no longer possible with 2.0.0

tobozo, 2018-04-24:
[looking](https://github.com/samdenty99/Wi-PWN/blob/40d137e43036c7f0765c3cfbdb28d7c9c3134031/arduino/Wi-PWN/Wi-PWN.ino#L165) at the [differences](https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/wifi.h#L256) that shouldn't be hard to add
@spacehuhn are you on it or do you need a PR?

spacehuhn, 2018-04-29:
This isn't a priority for me so it has to wait.
It requires adding additional settings and commands. Not diffucult but annoying to add. 

For a quick fix, you could just hard code the connection in the setup() function and compile it. I'm sure there are some examples on how to connect the esp8266 to an access point.

tobozo, 2018-05-12:
flagging this as [wontfix] *but* keeping the door open to a WiFi client configurator loaded in the context of the [sd-updatable](https://github.com/spacehuhn/esp8266_deauther/tree/sd-updatable) branch

spacehuhn, 2018-10-15:
Closing the door now, sorry. I don't have the time anymore for the changes that would be required by such a feature. Everyone is of course welcome to add the feature themselves.

closed 2018-10-15 by spacehuhn