SpacehuhnTech/esp8266_deauther #820: Steps to boot bin to esp32?

anupamb062, 2018-04-23:
I have tried to upload the 4MB and 1MB bins to esp32 using esptool.py but to no use. The file is uploaded but the program does not seem to work. I am a newbie to the esp32 based dev kit. Any way to port it to esp32? Will go through all he documentation but before that got too excited and wanted to try it out. Any leads would be appreciated.

Command that I use: esptool.py -c esp32 -p COM3 -b 115200 write_flash 0x000000 esp_deauther.bin

tobozo, 2018-04-23:
ESP8266 binaries aren't meant to work on ESP32, funny you could upload it though.

spacehuhn, 2018-04-24:
> Any way to port it to esp32? 

nope

https://github.com/spacehuhn/esp8266_deauther/wiki/Supported-Devices

closed 2018-04-24 by spacehuhn

anupamb062, 2018-04-24:
Hi @spacehuhn ,
                            It would be great if there would be a deauther repo for esp32 as well. I request you to please do a deauther project for esp32 as well. I am sure that I among many others would be pleased and would appreciate it a lot.

tobozo, 2018-04-24:
you can already find esp32 deauther repos on github, all you have to do is search

none of them does work for deauthing though, QED

spacehuhn, 2021-12-12:
Hi! There are many issues on this repo about ESP32 support, so I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want. I appreciate your understanding.