SpacehuhnTech/esp8266_deauther #821: Access deauther from raspberrypi?

darkprince1337, 2018-04-24:
Hello!
Can we access deauther from raspberry pi via MQTT server or something?

i'm trying to make a wireless intrusion prevention system (WIPS), which detects wireless attacks and block/deauth them and this is a hardware based project.
I'm able to detect wireless attack through suricata (Opensource IDS) which is installed in raspberry pi. The problem is , how to send input from raspberrypi to nodemcu to block/deauth the attacker?
PS:

tobozo, 2018-04-24:
You can easily achieve that using [this hat](https://shop.pimoroni.com/products/esp8266-phat), you'll have to add you own buttons and OLED and talk to the deauther through the serial port,

tobozo, 2018-04-29:
@darkprince1337 did you manage to talk to the deauther from your rpi? If so can we close this issue?

tobozo, 2018-05-09:
closing this as inactive, please reopen when something new comes up

closed 2018-05-09 by tobozo