SpacehuhnTech/esp8266_deauther #823: Help please, I'm a newbie

spider222, 2018-04-24:
sketch\Scan.cpp: In member function 'void Scan::start(uint8_t, uint32_t, uint8_t, uint32_t, bool, uint8_t)':

Scan.cpp:76: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'

     WiFi.scanNetworks(true, true);

                                 ^

sketch\Scan.cpp:76:33: note: candidate is:

In file included from sketch\Accesspoints.h:5:0,

                 from sketch\Scan.h:5,

                 from sketch\Scan.cpp:1:

C:\Users\Milos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\Milos\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool)'

I had other "issues" and I resolved them, but I just can't find any solution to this one. 
I'm a newbie so any detailed help would be most appreciated.

spider222, 2018-04-24:
I then checked the instructions - Select your board at Tools -> Board and be sure it is at ESP8266 Deauther Modules (and not at ESP8266 Modules)!  

I switched the module and I get to the very end and get this issue:

Archiving built core (caching) in: C:\Users\Milos\AppData\Local\Temp\arduino_cache_87468\core\core_deauther_esp8266_nodemcuv2_CpuFrequency_160,UploadTool_esptool,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_1M256_3cbd75fb0b0e9b1df1df88bfd938b71e.a
Sketch uses 683246 bytes (89%) of program storage space. Maximum is 761840 bytes.
Global variables use 42204 bytes (51%) of dynamic memory, leaving 39716 bytes for local variables. Maximum is 81920 bytes.
error: Failed to open COM1
error: espcomm_open failed
error: espcomm_open failed

spider222, 2018-04-24:
And, I'm not sure if I did this Flashing right, I did all with MCUFLASHER, had that 1MB bin file on my desktop, I flashed it and that's all. 

1st I followed some instructions from a YT video, where guy never switched module or even installed that 2nd board. Or even added that 2nd link in "Preferences"(had a lot of problems) And I did that now, and I get that message above at the very end. 

Any help guys? I'd really appreciate it :/

spider222, 2018-04-24:
I changed the PORT, and it's done!! 

Now I need to try it out only and that's it.

tobozo, 2018-04-24:
nice turing path, thanks for sharing :+1:

spider222, 2018-04-24:
Well now I need a password for this device right? SSID and a PASSWORD. How to find that?

spider222, 2018-04-24:
Scan for WiFi networks.
Connect to pwned with the password deauther.
Open 192.168.4.1 (or deauth.me)

Solution I suppose! 

Only one question, can I change name/password?

xXCrispyCatXx, 2018-04-24:
yeah, in the settings inside the web interface

tobozo, 2018-04-25:
I'll close this non-issue as self-solved, but it was very interesting to see how fast those most common know problems were solved and will be very helpful when it comes to optimize the documentation

thanks!

closed 2018-04-25 by tobozo