SpacehuhnTech/esp8266_deauther #824: No SSID?

xXCrispyCatXx, 2018-04-24:
For some reason the SSID text in the settings just made itself say this: onchange='save("ssid",this.value)'
I can't access the network on my IPhone, the SSID dosen't even show up. I can see it on my chromebook (won't connect), and i can only run the attack on my mac. The SSID just shows up without a name. It's blank. It worked great for a while, but now it dosen't. I tried re-uploading the code, but it just kept all the old settings and unfortunately, this problem. Is there a way to directly change it in the code? Or is it something easier to fix? Would really appreciate some help, thanks!

tobozo, 2018-04-25:
you may want to try the [reset sketch](https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch) to get things back to factory defaults

xXCrispyCatXx, 2018-04-25:
Oh, i didn't even see that XD Thanks for the help, this really is an awesome project. Keep it up!

closed 2018-04-25 by tobozo