SpacehuhnTech/esp8266_deauther #826: Wemos® D1 Esp-Wroom-02

zangaby, 2018-04-26:
Hello,

I have this board [https://www.banggood.com/Wemos-D1-Esp-Wroom-02-Motherboard-ESP8266-Mini-WiFi-NodeMCU-Module-p-1224577.html](url) which was working great on version 1.6, now I tried to flash the new version and even though the web interface is working, unfortunately the OLED display no longer works.

I set USE_DISPLAY to true and DEAUTHER_DISPLAY SSD1306Wire display but no luck.

Any thoughts on what I could be missing ?
Thanks !

tobozo, 2018-04-29:
Please post the different approaches you've tried to make it work i.e. try all config files one after another and note the results.
Also see what board you've selected from the Arduino tools menu and maybe try another one?

zangaby, 2018-05-02:
This is what I tried so far
WemosD1 board
D1, D2
5, 4

Generic Board 8266
Getting error when trying to compile: Multiple libraries were found for "Adafruit_NeoPixel.h"

NodeMCU board
Same as with the WemosD1

tobozo, 2018-05-02:
The message "multiple libraries were found..." is not an error but a warning, but this board looks like a bad quality clone and from the comments there are a few pinout issues you need to solve first.

First get your OLED to work alone, without the deauther (e.g. run the demo sketch from the SSD1306 library) and when you have it working, note down the pinout it's using.

Also the output of this may change depending on the board you select:
```
Serial.print(D1); 
Serial.print(D2);
Serial.print(SDA);
Serial.print(SCL);
```

zangaby, 2018-05-03:
Thank you, I tried running the demo sketch, and it's working with
`SSD1306Wire  display(0x3c, D1, D2);`

The selected board was Wemos D1 mini.

On he serial I see: D1 5 D2 4 SDA 4 SCL 5

tobozo, 2018-05-03:
have you tried the DStike board from the boards menu?

zangaby, 2018-05-03:
Just tried now, still no display.

tobozo, 2018-05-03:
can you paste the content of your A_Config.h on pastebin, I'll tag this issue hoping someone with the same problem came up with a solution.

it really sounds like some pin assignment is messing it up so we'll see

spacehuhn, 2018-05-03:
Have you tried ```set display true``` and ```save``` over serial and restart the device ?

zangaby, 2018-05-04:
The A_Conifig.h I am currently using is this one:
[A_Conifig.h](https://pastebin.com/J4aDU1E7)

Will try to set it thru the serial as soon as I get home.

zangaby, 2018-05-04:
SUCCESS ! :+1: 

Thanks a lot !

closed 2018-05-08 by tobozo

mayurpatelgec, 2018-08-04:
hi  @tobozo ,

i am having the same issue. working with library sample code with D1 D2 but not getting display with your code changes... i have tried all the things tried above..

@zangaby just want to know that what made ur code working? what was the final solution?

mayurpatelgec, 2018-08-05:
@tobozo hi thanks fr reply...

i know the pinout 
// 5 -> SDA
// 4 -> SCL

sample 1:
------------------------
https://github.com/ThingPulse/esp8266-oled-ssd1306/blob/master/examples/SSD1306ClockDemo/SSD1306ClockDemo.ino
i am running code of sample weather station with display without any problem with same code like
`SSD1306Wire  display(0x3c, 5, 4);`

 sample 2:
------------------------
https://github.com/ThingPulse/esp8266-weather-station/blob/master/examples/WeatherStationDemoExtended/WeatherStationDemoExtended.ino

with code 
```
const int SDA_PIN = 5; 
const int SDC_PIN = 4;
```

but no luck when i try with your code... with 5 as SDA and 4 as SCL

tobozo, 2018-08-05:
@mayurpatelgec my code ? what sketch are you talking about ? this is @spacehuhn's repository, not mine :rofl: 

please make sure you don't use vague terms such as "no luck" or "i am having the same issue" as this can lead to confusion and it doesn't help you get the problem sorted.

having the same issue does not imply the same cause, and having no luck is subjective

zangaby, 2018-08-07:
For me ` set display true` and `save `over serial as spacehuhn suggested was the solution.

mayurpatelgec, 2018-08-07:
@zangaby  hi..

can u explain a bit more how can i save over serial?

zangaby, 2018-08-07:
You open the serial console in the Arduino IDE, you enter the set display command
`set display true` then you enter the `save ` command to save everything. Afterwards you restart the device.

mayurpatelgec, 2018-08-07:
ok thank u @zangaby ..!!
i will try and post the updates..