SpacehuhnTech/esp8266_deauther #827: How do I add a another option in the display menu?

Booooozo, 2018-04-26:
I love your deauther program. But I also want to add a another option in the display menu like:

SCAN
SELECT
ATTACK
PACKET MONITOR
CLOCK

I want to add another option in the display main menu called "CLOCK" and when the CLOCK is chosen it should the following [code](https://github.com/dmainmon/ESP8266-OLED-World-Clock-IoT/blob/master/ESP8266_OLED_Clock_IoT.ino). I have been trying to edit your code to fit my needs for like 2 hours but no success. Any Ideas? Thanks!

PS. I am a newbie and started learning Arduino only a few days ago.

tobozo, 2018-04-26:
Hiding a deauther in a clock is a neat idea, we even saw one [hidden in a sandal](https://www.hackster.io/58569/wi-fi-deauthing-sandals-540e3b).

However, if you only started a few days ago maybe it's better to first merge your clock with a less complex project based on SDK 2.0.0.

This deauther project is using an old version of the ESP8266 SDK and you will get many conflicts when using code from projects based on other SDK versions such as this clock code.

Also the deauther is missing a few components to get what you need. 
It doesn't have (yet?) a WiFi client to connect to a NTP server and it doesn't have a RTC module to keep track of time in an acurate fashion.

spacehuhn, 2018-04-26:
Yep was thinking of something that keeps track of the time but it would require a lot of efford for something that doesn't really matter for any of the deauthers functions. Unless you want to script something that is time based of course.

The display code is one big confusing class right now. It's not too complicated, you can add menu entries relatively easy, just look how other menu's are build int DisplayUI.cpp. 
I want to rewrite it to make a library out of it, but that requires a lot of time that I can't afford right now.

Booooozo, 2018-04-27:
@spacehuhn I tried editing the DisplayUI.cpp like this:

``` // MAIN MENU
  createMenu(&mainMenu, NULL, [this]() {
    addMenuNode(&mainMenu, D_SCAN, &scanMenu); /// SCAN
    addMenuNode(&mainMenu, D_SHOW, &showMenu); // SHOW
    addMenuNode(&mainMenu, D_ATTACK, &attackMenu);// ATTACK
    addMenuNode(&mainMenu, D_CLOCK, &clockMenu); // CLOCK
    addMenuNode(&mainMenu, D_PACKET_MONITOR, [this]() { // PACKET MONITOR
      scan.start(SCAN_MODE_SNIFFER, 0, SCAN_MODE_OFF, 0, false, wifi_channel);
      mode = SCREEN_MODE_PACKETMONITOR;
    });
  });

      Arduino throws back "D_CLOCK was not declared in this scope"

How do I resolve this?

reaper7, 2018-04-30:
I use something similar to display the battery status,
this item is not editable, it means you can not enter it
and is displayed on empty bottom line (only in situations where the menu items is no more than 4, like the main menu).

dirty, unfinished code (at the end of **void DisplayUI::drawMenu()**):

```cpp
  if (currentMenu->list->size() <= 4) {
    #define NSAMPLES    100
    #define VOLTNOBATT  0.7
    //int r1 = 340000;  //voltage divider r1
    //int r2 = 100000;  //voltage divider r2
    int tempvalue = 0;

    for(uint8_t i = 1; i <= NSAMPLES; i++ ) {
      tempvalue += analogRead(A0);
    }
  
    tempvalue = int(tempvalue/NSAMPLES);
  
    //float res = ((((tempvalue * 1000) / 1024) * (r1 + r2)) / r2) / 1000.0;
    //float res = (((tempvalue / 1.024) * (r1 + r2)) / r2) / 1000.0;
    float res = ((tempvalue / 1.024) * 4.4) / 1000.0;
    
    if (res > VOLTNOBATT) {
      drawString(0, (4) * 12, " ACCU ");
      drawString(44, (4) * 12, String(res,2));
    } else {
      drawString(0, (4) * 12, " ACCU -.--");
    }
  }
```

spacehuhn, 2018-10-15:
I added a fake-clock to the menu now. Most dev boards don't have a RTC, so the clock just generates a random time on startup and can be adjusted using the buttons. 
This should be sufficient enough as a disguise. 😃

closed 2018-10-15 by spacehuhn

Ch4ist0, 2019-06-13:
Hello, is it possible to change in the menu ESP8266 Deauther 2.1.0 instead of "clock" battery level options?