SpacehuhnTech/esp8266_deauther #828: Error when compiling deauther 1.4 version.

Booooozo, 2018-04-26:
So I downloaded the deauther 1.4 version to make some major changes . But when I compile without making any changes with  Arduino it throws back this error which I can't seem to fix.

```In file included from /Users/macbook/Documents/Arduino/libraries/esp8266-oled-ssd1306-master/src/SSD1306Wire.h:34:0,
                 from /Users/macbook/Documents/Arduino/libraries/esp8266-oled-ssd1306-master/src/SSD1306.h:33,
                 from /Users/macbook/Downloads/esp8266_deauther-v.1.6/esp8266_deauther/esp8266_deauther.ino:28:
/Users/macbook/Documents/Arduino/libraries/esp8266-oled-ssd1306-master/src/OLEDDisplay.h:201:10: error:   initializing argument 1 of 'void OLEDDisplay::setFont(const uint8_t*)' [-fpermissive]
     void setFont(const uint8_t *fontData);
          ^
exit status 1
invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]```

I even have all the latest libraries installed. Any ideas on how to fix this?

Thanks!

spacehuhn, 2018-04-26:
You need a different OLED library version: https://github.com/ThingPulse/esp8266-oled-ssd1306/releases

However, I can't tell you out of the head which version you need because you're using the 1.6 Deauther.

Booooozo, 2018-04-26:
Yeah sorry I ma actually using the 1.6 version not 1.4

Booooozo, 2018-04-26:
@spacehuhn Which version of the library should I use?

Booooozo, 2018-04-26:
Well I am lucky I just installed the 3.2.4 version of the library and it compiled with no problems. Thanks!

closed 2018-04-26 by Booooozo

tobozo, 2018-04-26:
1.6 was probably referring to the hardware version rather than the code version?

Booooozo, 2018-04-26:
@tobozo I was referring version 1.6 of your deauther program