SpacehuhnTech/esp8266_deauther #832: Doesn't work

SyncedCyberSoul, 2018-05-01:
I have two versions of the board. The larger one, and the small compact one. I believe both are the 4mb modules. I have flased the board with arduino, and upon plugging it into a power bank, it blinks both LEDs twice, once normally and then again very faintly. 
Afterwards there is no result. No wifi network is being broadcasted.
I've tried multiple settings in arduino. nothing works
Please help lmao

SyncedCyberSoul, 2018-05-01:
Yes, I typed status and tested chicken and indeed got serial output from the chip. 
It just doesn’t do da wifi :(

It’s a very nice chicken by the way.

spacehuhn, 2018-05-01:
Have you tried flashing the reset sketch?
https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch

ifengchao, 2018-05-03:
I got the same problem...
Here's the log from com6:

Mounting SPIFFS...OK
Switched to Channel 1
Settings loaded from /settings.json
Settings saved in /settings.json
Device names loaded from /names.json
SSIDs loaded from /ssids.json
Scan results saved in /scan.json
Serial interface enabled
Started AP
[WiFi] Path: '/web', Mode: 'AP', SSID: 'pwned', password: 'deauther', channel: '1', hidden: false, captive-portal: true
STARTED! \o/
v2.0.5
Executing /autostart.txt
Done executing script

I do have reset the sketch, the log:

  STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initialized

ifengchao, 2018-05-03:
My board is wemos D1 with 4MB flash.

spacehuhn, 2018-05-03:
Do other sketches work with the WiFi AP? Arduino has examples for that.

ifengchao, 2018-05-03:
it works well with v1.6 firmware.

tobozo, 2018-05-03:
`works with 1.6` && `won't work with 2.0.x` could it be pin assignment problems?

Neopixels I'm looking at you :eyes:

SyncedCyberSoul, 2018-05-03:
Sorry I haven’t been able to get back to you. My cat seems to have nabbed one of my boards, I have yet to find it. The one she did take is the one I was having issues with unfortunately. The small one you recommended works but once I log in it crashes :(

xTardguard4, 2018-05-06:
same problem here, i've wiped the firmware, reset the arduino code, and reflashed the board several times and the serial monitor shows that everything is working but still no "pwned" AP

spacehuhn, 2018-05-06:
Can someone here please share the serial output for the sysinfo command?

spacehuhn, 2018-05-07:
Open the [serial monitor](https://github.com/spacehuhn/esp8266_deauther/wiki/Serial) and do `set mac random` then `save` and restart the device. Someone contacted me per email with a similar issue and it fixed it for him. I suspect some suspicious ESP8266 modules or something with the old SDK.

xTardguard4, 2018-05-08:
another thing that might work is `< erase_flash>` in esptool and uploading the .ino and the .bin with dio mode,
and also make sure that your usb cable works

tobozo, 2018-05-12:
closing this as fixed

closed 2018-05-12 by tobozo