SpacehuhnTech/esp8266_deauther #835: dont working anyone.

russianhackerman, 2018-05-01:
sorry for my bad english
so bad. 
i started in v1.5. fast web no errors all work. super
1.6. very slow (hard?) web many web errors and disconnect while attack started. (cant stop attack in web)
2.0. cant compile first gen. network is constantly falling for no reason. very hot chipset. (hardware+network fail when hot). many times reconnect while not doing anything. 
2.xx last update download today (1 may 10pm). successfully compiled&flashed and... nothing. no wifi (cant see pwned) trash in serial at 115200. used binaries and compile arduino ide.
now i use v1.5. (sorry for google translate)
now used the same iron? (hardware?) same ide and same bin flasher. (my english very bad)
1.5 working very good now in same iron/hardware. problem in updates 1.6-last. every time it gets worse and worse.

tobozo, 2018-05-02:
You can reset your ESP by flashing [this](https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch).

rdavydov, 2018-05-02:
By "esp8266 generic/mcu-v3" you mean you have tried it only on nodemcu v3 or on both the nodemcu and a separate esp module (which one, 12f?)
---
Под "esp8266 generic/mcu-v3" ты подразумеваешь, что пробовал только на nodemcu v3 или на nodemcu и на отдельном модуле esp (на каком, 12f?)

rdavydov, 2018-05-02:
If said symptoms occur only on one nodemcu v3 board, it is obvious that your board is a faulty one.

Если указанные тобой симптомы возникают только на одной плате nodemcu v3, очевидно, что твоя плата неисправна.

Kili0n, 2018-05-06:
Hello, why does this error come out?
How to fix it?
[image]

spacehuhn, 2018-10-15:
https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ#timeout-error

closed 2018-10-15 by spacehuhn