SpacehuhnTech/esp8266_deauther #836: Problem

Simoz11, 2018-05-02:
Hi I'm using a nodemcu (This one) 
[image]
And after flashing the program on to it, a red light turns on and a blue light flashes once. 
I have tried with the windows flashing software and the Arduino software with on luck.
can anyone help me out (I'm new to this) 
Thanks Simon

ProkhorovTweets, 2018-05-03:
Same thing here, using this particular board(https://ru.aliexpress.com/item/1-esp8266-nodemcu-V3-Lua-WI-FI-32-USB-Flash/32837914547.html?spm=a2g0s.9042311.0.0.xSh732), I've tried to upload multiple ways and multiple types of firmwares, through Arduino IDE and esp-tool . I've tried NodeMCU, generic 8266 and even DSTRIKE version. But right after the upload, which looks like always successful - there is no life at all, no blinks, no network, no experiments

gaMingLT, 2018-05-04:
Hello, 
i might be havig the same isseu here. After you flash it with the bin fil. Do you still need to upload the source code with arduino to your node mcu?

tobozo, 2018-05-18:
@doxmg do you have another NodeMCU you can compare with?

If you can flash & run the esp8266_deauther 1.6 and not the 2.x then it's probably the NodeMCU module isn't of good quality enough to run this kind of edge code.

doxmg, 2018-05-18:
@tobozo I do not have another esp8266 however my friend bought the same one and said something about how his works great on 1.1, so should I try 1.6 or 1.1 or even something else similar?

tobozo, 2018-05-18:
yep, it's worth trying the earlier versions

doxmg, 2018-05-18:
@tobozo ok thank you I’ll reply with an update soon

doxmg, 2018-05-18:
So update: I switched over to 1.1 and I tried the 230400, 9600, and 115200 baudrates along with 80MHz and DIO and I tried each with 4 and 1 MBytes on the nodemcuflasher program, and it still won’t show pwned or any other network even though it says it’s done flashing!  I don’t know what is going on at all.

doxmg, 2018-05-19:
@tobozo i tried just about everything and I believe now that I bought a dud module and I am going to try buying a new one

Virtix21, 2018-05-26:
I'm also having this problem, sorta. the earlier versions worked fine, I CAN'T see the WiFi signal on my Phone (Google Pixel 2) but I CAN see it on my desktop with my Alpha WiFi adapter, my laptop also doesn't see it. Otherwise, it's fully functional using the Alpha.

ExploiTR, 2018-06-05:
Try by holding the flash button and flashing then

ksibln, 2018-06-16:
No WLAN here, too...

I have a "LoLin" "new NodeMcu V3" unit.
After flashing 2.0.5, no WLAN is available.

When I re-flash to 1.60, everything works fine again.

I tried this several times.

NIEIXIUIS, 2018-06-16:
I kinda fixed this issue by first ereasing all the data on the board, then running a wifi connect script and make it connect to my wifi. Then i connected to the hotspot it made called esp8266. Then i uploaded the v1.6 script via arduiono. (i also changed the ssid and password in the script before uploading. It worked fine for a day then it suddenly said no internet access when i connected from my phone and wouldn't connect. I did the same thing as i wrote and now it works again. Somone know why this is. Why doesn't that get fixed when repluggin it or reset it?

spacehuhn, 2018-10-15:
Does anyone's issues here persist?
I can't make out any cause without further information and probably investigation on your side.

closed 2020-06-16 by spacehuhn