SpacehuhnTech/esp8266_deauther #841: Issue compiling

Gamerein, 2018-05-03:
```
Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 80MHz, DIO, 115200, 512K (64K SPIFFS), ck"

Archiving built core (caching) in: C:\Users\Rein\AppData\Local\Temp\arduino_cache_489431\core\core_deauther_esp8266_generic_UploadTool_esptool,CpuFrequency_80,FlashFreq_80,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck_78280db881ba30f086bb8adc06b3831e.a
c:/users/rein/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rein\AppData\Local\Temp\arduino_build_647562/esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
https://hastebin.com/luvevalucu.hs < same with verbose output

spacehuhn, 2018-05-03:
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation#upload-settings

tobozo, 2018-05-03:
I hate how Arduino IDE is sneaky with this, the real information in this error message is hiding itself at the end of a very long line:

`section `.irom0.text' will not fit in region `irom0_0_seg'`

Gamerein, 2018-05-03:
I didn't notice that setting when I went over them.
You guys are awesome thanks

closed 2018-05-03 by Gamerein

Alansadanandan, 2020-10-11:
i am beginner please give a set by instruction for fixing this error . please

Bommlon, 2020-10-16:
All the information you should need is [here](https://github.com/SpacehuhnTech/esp8266_deauther/wiki/Installation#installation).

olek07, 2020-11-08:
> i am beginner please give a set by instruction for fixing this error . please

Hello!
Didi you resolved the issue? I have the same problem