SpacehuhnTech/esp8266_deauther #842: Multiple errors

stephan2002, 2018-05-03:
> Please search for existing (open and closed) issues first to avoid duplicates.  
Also have a look at the [Wiki](https://github.com/spacehuhn/esp8266_deauther/wiki).  
Please put error messages and code ```inside these 3 quotes/grave accents```
(Delete this text when you've read it)

Hi, I have issues with the web interface...
When I power on my nodemcu a network ssid called "pwned" shows up.

I am connecting to it with the default password and when I open the web interface I accept the disclaimer. But when I try to search for APs it gives me the error: ERROR: timeout loading file run?cmd=save names

and sometimes it gives this error instead of the first one: ERROR: timeout loading file run?cmd=scan aps -ch all

When I reload the page all works fine and the access points are there.

When I select one and go to the attack tab and select one of the attacks the attacks are working fine.
When I try to stop the attack it is not working. After some waiting I  got error: ERROR: timeout loading file attack.json

When I try to access one of the webpages of the deauther they are just not responding.

I don't think this is normal. 

How can I fix this?

Stephan2002

stephan2002, 2018-05-03:
Oh and sometimes ERROR: loading file: run?cmd=attack      is showing up when I try to stop an attack

tobozo, 2018-05-03:
Your phone gets disconnected by the deauther during the channel hopping, this is perfectly normal.

With a serial console connected to the ESP you'd have more debug information.

closed 2018-05-03 by tobozo

stephan2002, 2018-05-03:
@tobozo but why is the deauther disconnecting me when I try to stop the attack? And why can't I access the webpages anymore when I stop the attack?

stephan2002, 2018-05-03:
When I try to reload the settings page it says: ERROR: timeout loading file settings.json

tobozo, 2018-05-03:
don't do a full channel scan, just scan the channel 1 and you won't get disconnected

BravadoRL, 2018-05-19:
I'm having the same problem. I can connect to the IP and also agreed to the terms.
Then I get to the dashboard where I can do the scan. I have set the scan to 1 channel as suggested but it still gives me the timeout error. @stephan2002 Did you figured it out yet?
Also @tobozo any suggestion on what I should do?

PS, I have flashed the board to the latest version FYI.

stephan2002, 2018-05-19:
No unfortunately not