SpacehuhnTech/esp8266_deauther #843: Exit status 1 Error!

gaMingLT, 2018-05-03:
Hello,
First of al i have closed my previous isseu "Thread". Sjo i tried everything again on a other pc, Latop in this case. I flashes the node mcu esp8266 as said in the guide. Then i followed the guide that you have to open en download some stuff in arduino, i did al that. I checked the upload settings, ect, ect. But again i got the same error. Exit status 1. ......h is missing, no such fil or directory. The weird thing is that tried different version of the deauther project and i get the same but exit status 1 error but a different ....h missing alert. But what al these again have in common i only see to find is that the #include <EEPROM.h> is missing. Anybody than can help? Would this be easier if i connected the node mcu esp8266 directly to a phone? 
Thanks in advance.
Beginner level
*srry for bad english

tobozo, 2018-05-03:
we urgently need this 

[image]

spacehuhn, 2018-05-04:
Bonvenon al la Esperanto Issue Friday (EIF). Hodiaŭ mi estos via gvidilo.
Unue vi devas specifi, kio ```...... .h file``` estas ĉar ni ne havas ideon.
Kion do la EEPROM-biblioteko devas fari kun ĉi tio? Ĝi ne estas uzata en la versio 2.0.
Kaj kion vi volas diri konektante ĝin al via telefono? Que?

gaMingLT, 2018-05-04:
Hello, 
Are you trolling me? Because i do not understand the language you typed. Pls translate or respond in english pls.

tobozo, 2018-05-08:
[this may be of some help](https://translate.google.com/?channel=fs&oe=utf-8&um=1&ie=UTF-8&hl=en&client=tw-ob#eo/en/)

tobozo, 2018-05-12:
@gaMingLT yes you need to improve the way you write issues otherwise you will get trolled :trollface: 

Please edit your issue so it is readable and detailed enough for us to identify and understand the problem.

- `download some stuff in arduino` << is not detailed enough
- `Exit status 1. ......h is missing` << why did you edit out the only interesting information of this message?

You approach was speculative, which is a good thing when experimenting, but this looks more like a lack of experience with the Arduino IDE and you're the only one capable of fixing that by practicing on different projects.

I'm closing this issue, if you feel you need to reopen it, please make sure you edit your initial comment first so we can be of some help.

closed 2018-05-12 by tobozo