SpacehuhnTech/esp8266_deauther #846: The state of the spaghetti code

spacehuhn, 2018-05-06:
Hey,  

I know my code isn't the greatest and that's ok. I had to make many little sacrifices to get stuff going. I haven't had the time to refactor code that really needed it (looking at you serial-interface, display-ui and LEDs!). I also learned a lot of better coding techniques after I published the version 2.  

Now I started making a library for the Command-Line-Interface (CLI) to clean things up.
I made a great lightweight library that I will continue to work on, release when it's done and use in future projects.  

However, I will keep the current way the serial interface is written.  
**Why?**  
Because changing it would make things even worse!

Right now it's messy but it works! It works great in terms of performance and memory usage. Adding this library will make the code look a tiny bit better but add a lot of overhead. Not to mention the countless hours I would spend refactoring the code.

C++ is messy and complicated sometimes.
I'm making this issue to keep you up to day on what I'm trying to improve and what not. I will keep a lot of the code as dirty as it is, because never change a running system! But that doesn't mean I'm not learning from my mistakes.

I will write another post when I'm starting with the display UI and LED classes.

Cheers.

spacehuhn, 2018-06-05:
Alright, so I have rewritten the LED classes to make them much more readable and easy to extend for any type of LED!
That also simplified the config file a bit.

I wanted to sort some things in sub folders and make the structure more clean but the c++ compiler from that SDK version doesn't like subfolders at all! Not sure why, maybe the compiler flags can be tweaked but I didn't want to mess around with that. (It would also require a SDK update and I'm sure most people won't manage to do that and open a bunch of new issues! So better we keep it as it is.)

Next up in my to-refactor list are the buttons and then the display :)

tobozo, 2018-06-05:
Found out the hard way that .cpp, .h and .c files don't have the same precedence order for includes, and libraries have an extended search path while .ino files don't, of course this changes along the Arduino IDE versions so there's no chance you can keep your sanity all along :] 

Maybe it's worth trying to package the deauther as a library?

```
#include <deauther.h>
Deauther deauther(OLED_ON);

void setup() {
  deauther.setup();
}

void loop() {
  while(deauther.available()) {
    deauther.loop();
  }
}

```

spacehuhn, 2018-06-05:
Yeah making it a library would make a lot of things easier and more clean but creates new problems too. I don't even want to think about touching the installation steps! As you said, there's no way to keep your sanity here :D

Anyhow, I managed to find a better solution to deal with all those LED classes. I made them inner classes, so everything about the LEDs is in the LEDController class. Proper object oriented and encapsulated :)

I will clean up the Scan and its related classes now.

spacehuhn, 2018-06-09:
Ok so before I could clean up the Scan stuff, I had to clean up the mess called `functions.h`, `language.h` and `wifi.h`. THe ESP8266 SDK 2.0.0 has problems with writing singleton classes and putting stuff in namespaces also generated huge problems, so I had to keep everything as global variables and functions 😕   
But I split all those global functions in mutliple files and made those files encapsulated so they don't have any dependencies and can be easily copied and used for other projects as well! 🎊   

Now I'm rewriting the Scan related classes and wow I will have to refactor everything!  

So the next version will be version 2.5 with a completly restructured code base. That will help me a lot finding bugs and coding new projects. Because good object oriented code can be reused and right now, I could not take the deauther code and use it for the PacketMonitor32, but when I'm done refactoring, I will! I already noticed that when I was rewriting the BeaconSpam repo, just taking the deauther code does not work, there are too many dependencies deeply in the code!  

Because of that, I will also release some new libraries that can be used for any Arduino project 😃

tobozo, 2018-06-09:
These are the two fields I've been researching until now:

1) Make the binary loadable from a SD Card + able to load a menu
2) Headless/Serial-Only (i.e. deauther as a M5Stack/Rpi shield)

Dunno what to expect of the upcoming changes but it'll be interesting to see if more or less difficult to integrate.

I haven't found a clean way to do part 1) but we both know it works so this can wait.
Couldn't go much far with part 2) either as there are too many output formats (debug, text, formatted text, json), but I'll keep on researching as this can unlock many external uses such as nodejs app, Android app, python app, etc.

I'm not sure of the path I should follow though ...  should I use separate print functions i.e. prnt_debug, prnt_text, prnt_formatted and prnt_json, and maybe just substituting Serial() object with CustomSerial() so it becomes easy to manage an output buffer ?

spacehuhn, 2018-06-09:
Good point. The headless deauther should be easier to do with the changes I'm doing right now. 
I have to think about this a bit...

spacehuhn, 2018-06-22:
Little update on my whole *trying to fix the code mess* story:
I'm rewriting everything again :D 
Ok not from scratch, it's more refactoring really but I'm changing so much, it's an entirely different codebase with just some similar variables.

And no I didn't go mad! (yet)
I'm doing this to become better with the more advanced projects I want to start. This codebase here is a perfect learning ground.

The Deauther version 3 will enable a lot of features I wanted to add in version 2 but didn't have the time to add. With the next version I will try to go full object oriented to make as much modular as possible. Making it easier to expand the hardware compatibilities, for example.  

I already made 3 new libraries for it:
- [SimpleList](https://github.com/spacehuhn/SimpleList)
- [SimpleMap](https://github.com/spacehuhn/SimpleMap)
- SimpleCLI (not fully debugged yet)

I called them simple because I thought that's what they are, but thinking more about it, they are pretty advanced for the average Arduino user. I would recommend you to use  the C++ standard library for lists, maps and other data structures. Unfortunately you can't always use those, in that case look at SimpleList and SimpleMap.

SimpleCLI is my command line parser. I'm still debugging and improving some things. It's also pretty advanced and can do a lot while maintaining a minimal memory footprint.

spacehuhn, 2018-10-10:
Hey it's been a while since I posted an update!  

I released the [SimpleCLI](https://github.com/spacehuhn/SimpleCLI) library and I'm working on a library to easily create display Menu's and one to easily manage and store different setting values.   

I was going to make a Deauther v3.0 with a rewritten base and a lot of improved features, but I realised that I couldn't fit the new code base in the little amount of RAM.  
So I was thinking of making two versions. One for the OLED and one for the web interface, but that didn't work out either.

I don't think it's a bad idea to make a stripped down web optimized version, however it wouldn't really  have much performance improvements.

I was just working on finishing the OLED menu and release the new version later this month, when I saw that the RAM was again completely filled up - and I already threw out the web interface!
I noticed that I hit a dead end on this project.
It's less my code and more the compiler that fucks it up. 
If you use more low level C code, the compiler is more friendly to you and I'm sure there's still a lot you could add to this project. But it would just become a bigger clusterfuck! 
I wish I could use more modern C++ features like with the ESP32, but I had to work around a lot of little issues. The project became too big and a pain to work on with those restrictions.  

**Since deauthing only works with the old SDK and there's no easy way to hack the newer SDKs, I decided to stop working on the big update.**

Most of the progress I made so far will go into a new PacketMonitor32 version and other projects. That's why I made everything into a library 😉 .

The next days I will update some smaller things and make a version 2.1. This will probably be the last major release for this project.

spacehuhn, 2020-06-16:
Long time, no read.

I've been working on [version 3.0.0](https://github.com/SpacehuhnTech/esp8266_deauther/tree/v3).
It's serial CLI only at this point but has a much cleaner code base and more advanced features.
We're also working on the [Huhnitor](https://github.com/spacehuhntech/huhnitor) for easier access to a serial terminal.
Documentation will be available under [spacehuhn.wiki](https://spaecehuhn.wiki) at some point of time. 
The plan is to use the wiki page for all major projects and versions, but it's all WIP at this point of time.

Anyway, as I'm going through all github issues to label, close or delete them, that's it for this issue too.
I don't plan documenting my development process further over this way. 
Maybe I'll come up with something better in the future, we'll see.

Thanks for your interest though :)
Hope this has been informative for someone!

closed 2020-06-16 by spacehuhn