SpacehuhnTech/esp8266_deauther #850: Connecting and disconnecting

KMRDNS, 2018-05-09:
After setup got ESP_1FF828 network but i can only connect for 1 sec then disconnect and again connect 
Thats whats i get in Serial Monitor

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

STARTING...
EEPROM initialized
EEPROM cleaned
SPIFFS initialized
SPIFFS cleaned
WiFi credentials erased
DONE!

spacehuhn, 2018-05-09:
What you posted is the serial output from the reset sketch. Can't see anything wrong with that.

KMRDNS, 2018-05-09:
Im noob in this theme sorry( But any device can't connect to ESP_1FF828 network, instant kick

Kleinhirn, 2018-05-10:
Hi,

i have the same problem.
On my CH34x device (AZDelivery NodeMCU Lua Lolin V3 Module ESP8266 ESP-12E) it works fine, but on my CP210x device (AZDelivery NodeMCU Lua Amica Module V2 ESP8266 ESP-12E) i can only connect for seconds bevore i get disconnected.
I do not know why it makes reboot all the time....
On serial monitor i get this:

Soft WDT reset

ctx: cont 
sp: 3fff4f80 end: 3fff52c0 offset: 01b0

>>>stack>>>
3fff5130:  0000001c 4000444e 3fff51c0 00000002  
3fff5140:  0000001e 4000422e 60000200 3ffeffb4  
3fff5150:  00000100 40004ac0 00000100 003fe000  
3fff5160:  00000100 3ffefed4 3fffc718 003fe000  
3fff5170:  00000378 4000410f 40101528 00000378  
3fff5180:  3ffefed4 000003fe 3ffefed4 3fff8838  
3fff5190:  3fffc718 3ffefed4 00000378 00000000  
3fff51a0:  003fe000 40220197 00000378 3fff8838  
3fff51b0:  00001000 40220221 3ffefed4 000003ff  
3fff51c0:  000003fd 3fff4298 003fe000 3ffefed4  
3fff51d0:  ffffff01 55aa55aa 0000001c 0000001c  
3fff51e0:  0000001c 000000f7 000000f4 000003ff  
3fff51f0:  402200c8 00000003 3fff4064 3fff4064  
3fff5200:  00000001 402200f6 00000003 3fff4064  
3fff5210:  4020f994 3fff4064 3fff111c 3fff4064  
3fff5220:  4020f9c2 3fff111c 3fff4064 4020f9e8  
3fff5230:  00000003 3fff111c 3fff41d4 402153c4  
3fff5240:  3fff4108 3fff111c 3fff41d4 4020e90f  
3fff5250:  3fff41d4 3fff4298 40213d04 3fff42a0  
3fff5260:  402012e6 000001f4 000001f4 3fff4298  
3fff5270:  3fff41d4 3fff111c 3fff1048 4020ed93  
3fff5280:  00000000 00000000 00000000 00000000  
3fff5290:  00000000 00000000 00000000 3fff4298  
3fff52a0:  3fffdc20 00000000 3fff4291 40213d42  
3fff52b0:  00000000 00000000 3fff42a0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

ERROR: Checknum mismatch! Current checknum=16; stored checknum=255

reset() checknum=16, EEPROM=255
reset settings...Reset complete!

Settings:
  SSID            : 'Wi-PWN'			|  characters='6'		|  hidden='0'		|  channel='1'		|
  Password        : ''		|  characters='0'		|
  Scan            : hidden-networks='1'		|  client-scan-time='15'	|
  Attack          : timeout='0'			|  packet_rate='10'		|  deauth_reason='1'
  Interface       : dark-mode='0'		|  new-user='1'			|  cache='1 (450s)'
  LED Indicator   : enable='1'			|  pin='2'			|
  MAC AP          : default='de:4f:22:18:d9:d1'	|  saved='de:4f:22:18:d9:d1'	|  random='0'
  Beacons         : mac-change-interval='1'	|  1s-interval='0
  Deauth Detector : all-channels='1'		|  channel='1'			|  alert-pin='2'	|  invert-pin='1'	|  scan-time='200'
  Other           : channel-hopping='0'		|  multiple-aps='1'		|  multiple-attacks='1'
  PIN Control     : state='000000'		|  names='Pin 3;Pin 4;Pin 5;Pin 6;Pin 7;Pin 8'

settings saved!

Settings:
  SSID            : 'Wi-PWN'			|  characters='6'		|  hidden='0'		|  channel='1'		|
  Password        : ''		|  characters='0'		|
  Scan            : hidden-networks='1'		|  client-scan-time='15'	|
  Attack          : timeout='0'			|  packet_rate='10'		|  deauth_reason='1'
  Interface       : dark-mode='0'		|  new-user='1'			|  cache='1 (450s)'
  LED Indicator   : enable='1'			|  pin='2'			|
  MAC AP          : default='de:4f:22:18:d9:d1'	|  saved='de:4f:22:18:d9:d1'	|  random='0'
  Beacons         : mac-change-interval='1'	|  1s-interval='0
  Deauth Detector : all-channels='1'		|  channel='1'			|  alert-pin='2'	|  invert-pin='1'	|  scan-time='200'
  Other           : channel-hopping='0'		|  multiple-aps='1'		|  multiple-attacks='1'
  PIN Control     : state='000000'		|  names='Pin 3;Pin 4;Pin 5;Pin 6;Pin 7;Pin 8'

Trying to sync the MAC addr with settings
Could not sync because settings are not loaded !
nameList cleared
Attack LED : OFF

Starting WiFi AP:

Soft WDT reset

spacehuhn, 2018-06-28:
To use the deauther software, install it by following the instructions
https://github.com/spacehuhn/esp8266_deauther/wiki/Installation

What you did is flash he reset sketch to erase everything from the chip. 

:)

closed 2018-10-15 by spacehuhn